Ultimaker/Cura #11590: Vase mode progressive flow on ramp up

colin-donar, 2022-03-02:
### Is your feature request related to a problem?

After the base layer for vase mode parts, I've noticed that the flow is not altered to accommodate for the start of the spiral. Because it starts at a 0mm layer height and progresses up to the full height during the first lap, I have been getting over extrusion on this first spiral layer. 

I've checked the gcode and sliced a simple cube using relative extrusion to highlight this. As you can see line 42 and line 47 have the same extrusion amount but line 47's layer height is at most 1/4 the full height.
[image]
Initial layer is 3.5mm with 3mm everywhere else.

### Describe the solution you'd like

Ideally this could be compensated for in the slicing for that first layer of the spiral.

### Describe alternatives you've considered

I've considered writing a plugin to calculate a multiplier for the extrusion value based off the given Z height value and the layer height in a line. However I don't have the knowhow or time right now...

### Affected users and/or printers

This probably only effects large format machine like mine. I currently running a 5mm nozzle on a Dyze Designs Pulsar extruder.

colin-donar, 2022-03-02:
[SpiralCubeShowcase.zip](https://github.com/Ultimaker/Cura/files/8173556/SpiralCubeShowcase.zip)
Project and gcode

GregValiant, 2022-03-03:
Good one.
Your cube can only have (4) Z changes per lap and I thought that was a bit simple.
I loaded a 1000mm diameter cylinder to your printer.  There were 182 Z changes per lap.  The flow looked to be a definite problem.  The first spiral move of the cube should have been at 25% flow.  The first spiral move of the cylinder should have been 0.549% flow.
With a 200mm diameter cylinder in my Ender there were also 182 Z changes in the first spiral lap.  I set up for relative extrusion to make it easier to read.
;LAYER:0
G1 F600 Z0.2
G1 F2100 E5
G1 F3000 X214.8 Y115 **E0.12471**  ;1st extrusion at .2mm layer height
When spiral starts:
M104 S210
G1 X214.8 Y115 Z0.201 **E0.12471**  ;Same E with an actual layer height of .001mm
That E should have been 100%/182 * .12471 or .00068 ?
It's possible that [This Post](https://community.ultimaker.com/topic/23966-vase-mode-solid-bottom-offset/#comment-216130) on the Ultimaker Forum may be related as it is also with a large bore nozzle.
The condition seems to occur whether there are bottom layers or not, the first lap of the spiral still shows the issue.

smartavionics, 2022-03-03:
See https://github.com/smartavionics/Cura/releases for a cura variant that does what you're asking for. It tapers the flow at both the first and last spiralized layers.

smartavionics, 2022-03-03:
Oh, yeah. It can also modulate the spiralized wall width if you have a suitable model.

smartavionics, 2022-03-03:
Yeah, something's not right there. It's late now here so I'll take a look tomorrow and get back to you.

smartavionics, 2022-03-04:
Hi, yes, some regression had occurred(!) and the current release isn't working as it should with regard to the tapering of the flow of the bottom layer of the spiralized wall. Sorry about that, I will make a new release soon.

GregValiant, 2022-03-04:
@smartavionics this is from the OP's original gcode file:
;LAYER:0
M107
G1 F600 Z6.5
;MESH:10x10Cube.STL
G0 F3600 X340 Y805 Z6.5
;TYPE:WALL-OUTER
**G1 F600 Z3.5**
G1 F6000 E84.15034
G1 F480 X340 Y315 E6773.63438
G1 X830 Y315 E6773.63438
G1 X830 Y805 E6773.63438
G1 X340 Y805 E6773.63438
M104 S240
G1 X340 Y315 **Z4.375** E6773.64518
G1 X830 Y315 **Z5.25** E6773.64518
G1 X830 Y805 **Z6.125** E6773.64518
G1 X340 Y805 **Z7** E6773.64518
;TIME_ELAPSED:505.818539
;LAYER:1

and this is from the Gcode you posted:
;LAYER:0
M107
**G1 F600 Z6.5**
;MESH:10x10Cube.STL
G0 F3600 X340 Y805 **Z6.5**
;TYPE:WALL-OUTER
G1 F600 **Z4.375**
G1 F6000 E84.15034
G1 F480 X340 Y315 E2963.46504
G1 X830 Y315 **Z5.25** E4233.52824
G1 X830 Y805 **Z6.125** E5503.58671
G1 X340 Y805 **Z7** E6773.64518
M104 S240
;TIME_ELAPSED:260.764111
;LAYER:1

It looks as though the first loop at "Initial Layer Height" is missing and the "spiral initial layer" goes down at Z4.375 to Z7.0.  That seems to imply an under-extrusion?

smartavionics, 2022-03-04:
>  That seems to imply an under-extrusion?

Sure, that's because the first spiral layer actually starts at z = 0 so, in theory, you would start the spiral with zero flow and ramp the flow up to 100% by the end of the first layer. Of course, extruders don't work at 0% flow so I set a minimum of flow of 25% (I think it is) for the start of the first spiralized layer.

smartavionics, 2022-03-04:
Here's an image of another model that has a  lot more segments, you can see the smooth increase in flow on that first layer..
[image]

GregValiant, 2022-03-04:
I'm sorry to bother you with these questions but I'm trying to get my head around this.
_"because the first spiral layer actually starts at z = 0"_
I don't see that in the snippet.  It looks like the Z moves to Z6.5 (initial layer of 3.5 + zhop at 3.0) and then drops to 4.375 at the start of the spriral?  Because the initial flat layer isn't present in your gcode it would seem to imply that the first spiral layer is going down over air and has to take up the gap left by the missing flat layer?  That's the point I'm stuck on.

smartavionics, 2022-03-04:
Yeah, I just realised that I've goofed (yet again!) Will fix further...

smartavionics, 2022-03-04:
This should be better.

[CFFFP_SpiralCubeShowcase.zip](https://github.com/Ultimaker/Cura/files/8186206/CFFFP_SpiralCubeShowcase.zip)

Note that the preview display doesn't show the tapered layers but they are in the gcode.

GregValiant, 2022-03-04:
Before you ask...Yes, now I'm happy.
It's always nice to stick someone else with a bunch of work while I sit back and sip my coffee.

GregValiant, 2022-03-04:
I read gcodes into AutoCad so that I can see all the moves (G2 and G3 included).  It makes measurement and analysis much easier.
Instead of Cura's view of this Arcwelded gcode...
[image]

I see this...
[image]

smartavionics, 2022-03-04:
Hi, I made a new release, please give it a go.

All feedback is welcome but please do it via https://github.com/smartavionics/Cura/issues rather than here to avoid spamming the UM folks.

Ghostkeeper, 2022-03-14:
Forked slicers notwithstanding, this is also an issue in Ultimaker Cura. I could reproduce this as well. It could result in a bit of a rim on the print where one layer gets overextruded.