Ultimaker/Cura #11591: Unnecessary support generated

SkylinerXwiz, 2022-03-03:
### Application Version

4.13

### Platform

win 10

### Printer

Prusa

### Reproduction steps

slice a model

### Actual results

generate unnecessary support at the bottom and model abit offset from previe
[Desktop.zip](https://github.com/Ultimaker/Cura/files/8175459/Desktop.zip)
w

### Expected results

there shouldnt have support at the location, and the model edge should not be offset

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

nil

GregValiant, 2022-03-04:
The problem really is that there aren't any circles in an STL file.  Everything is composed of triangles and that makes for facets and straight lines.  Your Hole Horizontal Expansion of 0.5 is causing this behavior.
You will need to turn off Hole Horizontal Expansion or turn off support.  The lower large rectangular feature is a hole.  When the posts start then it isn't a hole anymore.  If you look you will see that Layer 13 has a step on the inside of each post and at the corner posts because Hole Horizontal Expansion has ended at layer 12.  Cura is picking up some of them as needing support because they hang over air.
In this image I have cranked HHE up to 1.0 (and turned off coasting) so you can see the situation better.  Cura has decided that a 1.0 overhang requires support on all of them.
[image]

This is with HHE turned off.  You can see that the seamless rectangle and the inside walls of the upright posts are now co-linear and the support has gone away.

[image]

@smartavionics has worked out a fix for this in his fork of Cura.  I don't know if it has made it into a released version yet or if there are plans at Ultimaker to incorporate his fix into the UM version of Cura.  The fix is to allow the user to enter a maximum "diameter" (periphery really) of a feature and so that rectangular hole would be ignored.

Ghostkeeper, 2022-03-14:
Looks like a duplicate of https://github.com/Ultimaker/Cura/issues/7487?

closed 2022-03-21 by no-response[bot]