Ultimaker/Cura #11598: cant slice my models ... nothing

Toddi3D, 2022-03-03:
### Application Version

Arachne 2 beta

### Platform

Windows 10 64

### Printer

Anycubic Vyper

### Reproduction steps

loading model but cant slice it, nothing

### Actual results

see above

### Expected results

nothing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8178779/cura.log)
[image]

GameKyuubi, 2022-03-05:
Maybe the same problem here #11580?

fvrmr, 2022-03-07:
Hi @Toddi3D thank you for your report.
Could you share your project file? (file - save project) You can share it in a .zip here on Github.

closed 2022-03-21 by no-response[bot]