Ultimaker/Cura #11600: Universal Settings do What?

inspectionsbybob, 2022-03-03:
### Is your feature request related to a problem?

It would be nice to know WHAT is being changed.  Some things set 77 settings, others as few as 23.  It would be nice to know what/why things are being set the way they are so we can tune for a Combination of issues....

### Describe the solution you'd like

Just a text file that says what is being set

### Describe alternatives you've considered

Trying to find a description on the website

### Affected users and/or printers

All

fieldOfView, 2022-03-03:
I think you are talking about the Universal Cura Settings plugin. That plugin is not maintained by Ultimaker, and has its own github issue queue here: https://github.com/5axes/UniversalCuraSettings/issues

inspectionsbybob, 2022-03-04:
Ah... I though that was a Cura product...
Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

closed 2022-03-06 by nallath