Ultimaker/Cura #11602: Whole Expansion creating changes in wall thickness

johnReeferVermont, 2022-03-03:
### Application Version

4.13.9

### Platform

Windows 10

### Printer

Creality Ender 5 plus

### Reproduction steps

1.- Create a box and shell it out making the walls 2 mm
2.- On one of the side walls, make a hole
3.- Export shape to Cura in STL Binary format
4.- Slice.
   4.1.- When you slice in cura the row above the hole is wider than the rest of the wall

### Actual results

The wall thickness changes when there is a value in the whole horizontal expansion 

### Expected results

If we use hole horizontal expansion it should not change the thickness of the wall.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

smartavionics, 2022-03-03:
Think about it, your box model is just a solid block with a large hole in the middle so the hole expansion will change the position of the inner wall. But the layers that have the hole in the side of the box no longer look like a block will a hole in a middle (when viewed from above) but, instead, look like this [_], i.e. there's no hole (when viewed from above) so the hole expansion is not applied to those layers.

johnReeferVermont, 2022-03-03:
Thank you for responding quickly.  Not sure I understand how would I then make a hole on the side that would have the expansion set to it.  Without the expansion, the holes are too small.

Thanks

smartavionics, 2022-03-03:
The hole expansion works in the x/y plane (vertical holes in a horizontal surface) not the z plane (horizontal holes in a vertical surface). So using hole expansion is not the right solution if the holes are horizontal (like you show in your image).

johnReeferVermont, 2022-03-03:
Sorry.  I don't understand.  I need to have holes on the bottom and the sides.  How do I do that if, when hole horizontal expansion is used, the slicer makes a mistake and changes the wall thickness.

Sent from Note 20 Ultra

smartavionics, 2022-03-03:
> the slicer makes a mistake and changes the wall thickness.

The slicer isn't making a mistake, you simply don't understand how the feature is intended to be used.

johnReeferVermont, 2022-03-03:
Ok.  Can you explain to me how I can print with holes on the bottom and on the sides.  I need the holes on the bottom to be adjusted by the setting and if I do, then it messes up holes in vertical plains.

Also, I am a solutions architect and can tell you that you should not be condescending.  If I don't understand and you brutally point it out without telling the client how to address it, seems to be unprofessional.

I want the holes, in the x and y planes to be right sized and also want the holes in the z access to n9t deform the shape.  If I don't understand,  pretend I am a customer who is reporting an issue and tell me how do I configure Cura to handle holes, which require adjustments, on the x and y plane and that the shape also has holes in other places.

Thanks

Sent from Note 20 Ultra

smartavionics, 2022-03-03:
Sorry I offended you.

smartavionics, 2022-03-03:
There is no viable solution provided by Cura at this time.

It is too difficult for me.

BTW, I'm not a UM employee, just a random person on the internet attempting to explain why the cura feature you have been exploring is not going to do what you would like.

johnReeferVermont, 2022-03-03:
Say no more.  That explains it.  Feel free to not help me in the future.  I may be posting more questions that are beyond your ability to be professional.  Wait, being professional would imply that you are a professional person.

I will be contacting github to let them know how pleasing and fulfilling was this interaction with you.   But then again, I have met many like you so I know your type and you will always think you are right and if someone knows less than you, you make a point in talking down to that person.  But I get it, you need to feel right all the time and you need to talk down to people.  That is how you live your miserable life.

fvrmr, 2022-03-07:
Hi @johnReeferVermont thank you for your report.
@smartavionics is right about this setting. 
See also my explanation of this setting here: https://github.com/Ultimaker/Cura/issues/9185#issuecomment-767543294 where a user had a similar question.
You can also read more about this setting in the settings guide: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/hole_xy_offset.md
If you want to change the values of the holes it is best to do this in a CAD program.

I will close this issue since it is a duplicate of: https://github.com/Ultimaker/Cura/issues/9185

closed 2022-03-07 by fvrmr

johnReeferVermont, 2022-03-07:
Thanks

Did you see how he treated me?.  Is this how your organization treats customers asking for assistance?.  This is my first experience with github and not a good one at all.

Please explain to me why he treated me the way he did?

Thanks

See also my explanation of this setting here: #9185 (comment)<https://github.com/Ultimaker/Cura/issues/9185#issuecomment-767543294> where a user had a similar question.