Ultimaker/Cura #11609: Merging does not merge the models and parts are missing

Minabe, 2022-03-05:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender-5 Plus

### Reproduction steps

1. Separate parts of the model
2. Load both parts of the model in Cura, enable "Fuzzy Skin" setting in the per-model tab (or not, same result), then merge them
3. Slice the model
[image]
[image]

### Actual results

When slicing not only does Cura separates them but most of the model isn't being sliced at all.
[image]
[image]
[image]
[image]

### Expected results

Model should merge and slice correctly.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Both parts have the same vertices where the walls meet as well as the same origin, no scale/manipulation was done after separating them and as such should merge correctly.
[Project_File.zip](https://github.com/Ultimaker/Cura/files/8189360/Project_File.zip)

GregValiant, 2022-03-05:
This is the repair report on the black part.

--> 317 Naked edges (?)
--> 3 Non-planar holes (?)
--> 12 Non-manifold edges (?)
--> 18 Inverted faces (?)

Here is the report on the fuzzy part:
--> 304 Naked edges (?)
--> 2 Non-planar holes (?)

When Cura gives a warning regarding a model then there is generally a problem with the model.
If a model is part of a group you can hold down the CTRL key when you click on it to select just the one part.
This 3mf has the models repaired.  Set Cura up and see how it goes.
[GV_CE5P_V3.zip](https://github.com/Ultimaker/Cura/files/8189617/GV_CE5P_V3.zip)

Minabe, 2022-03-05:
Sadly your "fix" destroys the model as you're not fixing the model as a whole but each part separated. It also doesn't explains why a part of the model does slice while the rest doesn't, even if incorrectly with the parts separated.

Playing with Cura's settings i got a similar result using "Keep disconnected faces" which make me think vertices aren't being overlapped and the "Union overlapping volumes" is not doing anything here. I have also increased the edges in order to overlap them even more but i always get the same result. I had read merging was a snap process in Cura and was hoping it would work that way but i don't know if it's bc Cura really needs manifold geometry or bc it's just how it works (bugged).

I was following this guide:
https://asherwinn.com/variable-fuzzy-skin-for-functional-printed-parts/

In there it says and i quote: "Best practice is to take full X-Y plane cuts. The reason for this is when combining later in Cura, if different bodies exist in the same Z layer, Cura will add internal walls to the print." Which is what seems to be happening here and what i was expecting but then i got most of the model not being sliced at all, there's definitely something wrong here and it's not the model's fault.

GregValiant, 2022-03-05:
As a secondary check I opened it in PrusaSlicer...
[image]

And as a tertiary check in IdeaMaker...
[image]

So I think that it might be possible that the errors within both models could conceivably, maybe, in some way, shape, or fashion, be causing a problem.  This is not to say that the repairs performed at [https://formware.co/OnlineStlRepair](https://formware.co/OnlineStlRepair) are correct although I think the term "destroys the model" might be a bit over-stated.

Back within Cura there are a lot of highlighted areas.
[image]

Minabe, 2022-03-05:
Those highlighted areas dissapear when the models are merged unless the merging was performed wrong (which would happen if the root position of the models is different, like resetting all transforms within Cura then trying to merge would result in a bad merge exactly like the result you got in IdeaMaker).  I don't get what does other slicers have to do with Cura tho, using Blender and a simple boolean operation i get the same model back so it does end up being about the way Cura merges, which i have not found much information about other than the tutorial link i provided.

Also, you're focusing on the wrong part, the main bug is that the models are being "merged" separated (walls are being added where the parts "merge"), that is most certainly a bug that affects the Z axis as it does not seem to happen with the X and Y axis.

GregValiant, 2022-03-06:
My point in bringing up the other slicers was that I couldn't get it to slice in any of them.  Besides PrusaSlicer and Ideamaker I tried with Matter Control, KissSlicer, QidiPrint, Pronterface, Mingda, and Chitubox. 
Here it is sliced in Qidiprint.  This looks familiar to me.
[image]

Cura certainly has bugs in it.  Whether this is one or not isn't my call but it would seem that if it is indeed a bug then a lot of other software's share the same bug OR the problems with the model are insurmountable by any of them.
The Cura team will be back to work on Monday and may address your bug report.  It is certainly repeatable with your model.

fvrmr, 2022-03-07:
Hi @Minabe thank you for your report.
But this is an issue with your model and not with Cura. The reason why this weird merging is happening is also because your models are not watertight. 
It also shows the highlighted areas for me when I merged the models:
[image]
So please make sure that your models are correct and watertight, so you can slice them as you would expect.

Ghostkeeper, 2022-03-20:
Cura treats each mesh as a separate model. These meshes individually are completely broken, with big openings making them not watertight. They have no volume or anything. And so Union Overlapping Volumes also has no effect; it only merges proper volumes. Cura doesn't determine the volume that these models fill.

To actually make Cura treat the two meshes as a single mesh, you could load up that 3MF file and export it as STL. That will combine the meshes into a single mesh, because the STL file format doesn't support grouping of submeshes. Loading that STL back in makes Cura treat it as a single mesh then.

However even then, Cura will complain that the resulting mesh is not watertight since the edges don't precisely meet up. Cura's stitching indeed snaps, but only by 5 micrometres (intended for rounding errors), and the gaps in the combined meshes are larger than that.

[image]

closed 2022-06-15 by nallath