Ultimaker/Cura #11610: custom selection mode won't work

akkos001, 2022-03-05:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

1. Opened "manage settings visibility"
2. Changed the visible settings for custom selection mode
3. Closed "manage settings visibility"
4. Changed to a different visibility mode (basic)
5. Clicked on custom selection

### Actual results

After I change back to the custom selection mode, the settings I had chosen get lost and it takes the settings visibility from the last mode I had. For example if I change from Basic to custom selection it will still show the basic settings visibility. Same goes for the other modes. The screenshots below show, that although I didn't change anything with the custom selection, it will just copy the mode I had activated before.
[image]
[image]
[image]

### Expected results

It should show me the settings I have chosen before and not the preconfigured modes.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

/

fvrmr, 2022-03-08:
Hi @akkos001 thank you for your report.
I can't reproduce your issue. Could you share a screen capture?
[image]

akkos001, 2022-03-09:
Hello @fvrmr, thank you too for trying to find the problem. 

I believe this bug was caused due to some plug-ins I had installed.Before, I had added many plugins, which sounded useful. There Probably was a problem with a plugin. After I made this post, I uninstalled cura and reinstalled the clean version of cura without any plugins. I couldn't reproduce the bug. So I made a backup and slowly added the plugins which I need / use frequently, but the bug hasn't appeared yet. I will observate cura a little longer and if the bug won't happen again, I guess I can call this issue as fixed.

fvrmr, 2022-03-15:
Thanks for your reply, I will close this issue. But if it happens again please let use know!

closed 2022-03-15 by fvrmr