Ultimaker/Cura #11612: Cura 4.13.1 fails to print vertical holes and strangely prints a halfway wedge in horizontal holes

paulbauwens, 2022-03-05:
### Application Version

Cura 4.13.1

### Platform

Windows 10

### Printer

Ender 3 + Marlin Bugfix may 2021 on  Ender 3 32bit 4.2.7 motherboard wiht 

### Reproduction steps

I am using Marlin Bugfix mat 2021 on an Ender3 4.2.7  32-bit motherboard with BLtouch no such issues before
I checked the STL files with Tinkercad, they look completely OK

Slicing and Printing STL files form this public source:

https://www.prusaprinters.org/prints/123038-table-clamp-for-miniature-vice-30-100-printed

file:  Table protector for clamp V3.0.stl  fails to print the hole in the bottom for the screw after Cura slicing
file:  Table clamp for miniature vice V3.0.stl   prints strange wedges halfway through the horizontal holes in all 4 holes
prints are otherwise completely smooth , 0.2 mm layer height
Using Grid infill pattern and 40% infill

[image]

P.S.  I re-sliced these files with Cure 4.12.1 and the problem does not appear, perfect usable prints

### Actual results

You can reproduce the wedge in de big hole of file  Table clamp for miniature vice V3.0.stl   This makes the print unusable.

The other three smaller holes in this object will also have a wedge halfway in the hole

### Expected results

The expected result was a hole without the wedge with a smooth thread for the accompanying printed screw.

The small nick in the photo was a result from my attempt to remove the wedge

Slicing and printing the Table protector for clamp V3.0.stl  shows a massive print without the hole in the bottom

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8190747/cura.log)
[cura_cfg.txt](https://github.com/Ultimaker/Cura/files/8190754/cura_cfg.txt)

GregValiant, 2022-03-05:
In Mesh Fixes -turn off "Make overhangs printable" and "Remove all holes".

paulbauwens, 2022-03-05:
Thanks, I will try this.  I can’t remember ever touching this setting, coukd it come from the defaults?
Regards, PaulBauwens

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

GregValiant, 2022-03-05:
They should have been turned off by default.  I have had a couple of occasions to use Remove all Holes.  The Make Overhangs Printable setting I have never used.  I do tell lot of people to turn it off though.  Without a better look see I'm guessing it's what caused a chamfer within your horizontal holes.

fvrmr, 2022-03-07:
Hi @paulbauwens thank you for your report.
Can you let us know if the suggestion of @GregValiant worked for you?

paulbauwens, 2022-03-07:
I will check and teport didnt have time yet

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

closed 2022-03-24 by no-response[bot]