Ultimaker/Cura #11613: Visual indication for userdefined seam position

BloodyRain2k, 2022-03-06:
### Is your feature request related to a problem?

There are rare cases where I prefer to manually tell Cura where to place the seam and the setting for that is already there.

What's missing though is being able to /see/ where "there" is.
There's the numbers for the position, but so far I've always needed to nudge my model around until the white dots actually started appearing where I wanted them.

### Describe the solution you'd like

The simplest way I could think of would be to "just" overlay a red and green line on the bottom grid which correlate with the userdefined coordinates.

Here's a mock up screenshot:
[image]

### Describe alternatives you've considered

As mentioned: nudging the model around until the result fits, because I simply can't /see/ the point on the grid from just looking at 2 numbers.

### Affected users and/or printers

.

fvrmr, 2022-03-10:
Hi @BloodyRain2k thank you for your request.
I will bring it up with the team. Keep you posted.

fvrmr, 2022-03-17:
I have discussed it with the team, and we decided to create a ticket for this on our backlog.
Devs see CURA-9056