Ultimaker/Cura #11616: One at the time GCODE is just all at once

bart989, 2022-03-06:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Loaded parts
2. Set multiple parts to print ine at the time in sequence
3. Preview Gcode Was looking okay
4. send to printer
5. Print was going all at once
6. Chcecked generated Gcode, it was "all at once" with 488 layers instead 2888 as in the preview
7. Generated other Gcode - same result, with preview looking good

### Actual results

Bad Gcode was generated, without function I was looking for

### Expected results

Gcode should've 2888 layers, and be printed one at the time

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura_gcode_bug report.zip](https://github.com/Ultimaker/Cura/files/8193296/Cura_gcode_bug.report.zip)

GregValiant, 2022-03-06:
That CE3PRO gcode file looks good to me.  There are 5 ";LAYER:0"'s in the file.  The term "LAYER:" appears in the file 2888 times.

Each part has it's own LAYER:0.  If one part ends on the completion of LAYER:488 and Cura moves to the next part it will be to print LAYER:0 of that part (and not LAYER:489).

When Previewing a gcode that is "One-at-a-Time" Cura can't show any difference because of all the duplicate layers in the file.  If you move the slider to LAYER:0 it shows ALL the LAYER:0's at once exactly the same as if they were print All at Once.

You will see in the gcode that the first part prints to layer 488 and at the end there is a layer 0 and the gcode continues with the skirt for the second part.

bart989, 2022-03-07:
You are right, I've noticed it today. Still when I open .GCODE in Cura I have problem seeing order of printing in preview. This is much less of an issue, but would be usefull. Should I create another bug, or maybe a feature request? Is this actually fixable?

GregValiant, 2022-03-07:
Either a Bug Report or Feature Request would work.  I have a feeling this has come up before and is something the Cura team may be aware of since it certainly "looks wrong".
Regarding the actual printing of the file - whether it prints from 0-250 and then 0-250 and another 0-250 or if it was to print 0-750 would not make a difference so long as the slice was right.  Having an Initial Layer Height, Line Width, etc. on layer 0 and layer 251 and layer 501 may be a show stopper.  That's conjecture on my part.
The ArcWelder plugin also has an issue when displaying the gcode file as Cura does not display G2/G3 commands.  Both of these Preview problems may be addressed when Cura 5.0 gets released.

fvrmr, 2022-03-15:
Hi @bart989 thank you for your report.
And thank you @GregValiant for the correct answer. 

About the g-code preview I can reproduce this, I will check with a colleague but I don't think that this will get a high priority to fix/implement. Since the normal preview is showing it correctly. 

Also the displaying of G2/G3 commands will not be implemented in Cura 5.0

fvrmr, 2022-03-21:
It has been discussed with the team and it has indeed no priority for us to fix or implement this, therefor we decided to defer this. I hope you understand.

carribeiro, 2022-06-30:
Hi, I had this same bug and while I understand your decision to defer, I'm thinking about a way to solve it. I'm not programming as I used to +10 years ago (I still use Python but just for quick scripts & stuff), so I can't make any promise but I think it would be a good intro do Cura code. If I feel it's a viable solution I'll let you know.

GregValiant, 2024-11-16:
I've been writing some post-processing scripts based on requests I've seen here and there (and on ideas I've had myself).
I have included some of the _Unlikely-to-be-needed-very-often_ scripts in a collection I call "Little Utilities".  One of them is "Renumber Layers" for One at a Time projects.  It will go through the Gcode and change the layers numbers to a conventional "All at Once" style from Layer:0 to the top layer of the file.  That was to provide more options for "Pause at Height" and Filament Change.  A side effect is that opening a renumbered Gcode in Cura will display properly as a One-at-a-Time" file in the preview.

Unzip the file and put "LittleUtilities_v15.py" into the "scripts" folder in your "Configuration Folder" it will be available with the other post processors.  After adding the script to your project - enable it and then select "Renumber Layers" and then "Renum>AllAtOnce". 
[LittleUtilities_v15.zip](https://github.com/user-attachments/files/17783255/LittleUtilities_v15.zip)

[image]

At some point I may put a pull request in for Little Utilities as it does include a couple of Cura bug fixes.