Ultimaker/Cura #11617: Unable to change infill density with support blocker

chasrobertson, 2022-03-06:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Weedo X40

### Reproduction steps

1 Used suppoirt blocker to surround area to have 100% infill. Original model was 30% infill with Cubic Subdivision.
2 Set infill density of support blocker (infill mesh only) to 100%.
Sliced model and only the walls are printed on the 100%  area.

### Actual results

Would work up to 40% infill, but after that only the walls are printed

### Expected results

Should have bee a solid area at 100% infill
[image]
.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Have tried it on different areas with the same results

smartavionics, 2022-03-06:
What happens if you use different infill patterns in the 100% region, i.e. lines or zig-zag?

chasrobertson, 2022-03-06:
Hi,
It is OK with lines and zigzag, and if I change it to this in the
eraser menu, where the rest of the model is in the other setting, it
works OK. Trust me to choose the setting which has problems!

Thanks again for getting back to me so quickly,

Cheers

Chas.Robertson.

>>> Mark Burton ***@***.***> 06/03/2022 21:56 >>>
What happens if you use different infill patterns in the 100% region,
i.e. lines or zig-zag?
—
(
https://github.com/Ultimaker/Cura/issues/11617#issuecomment-1060046519)
(
(
 or Android
(

chasrobertson, 2022-03-08:
Hi, I have an update. Printed with 100% infill in certain areas with setting of zigzag. Looked fine initially, but structurally unsound as the infill is separated from the outer wall by a small amount all the way around for the entire length. The gap is uniform on all sides so you end up with an outer wall, small gap and then the solid infill. Any ideas?RegardsChas.Robertson.Sent from my Galaxy<div>
</div><div>
</div><div>-------- Original message --------</div><div>From: Mark Burton ***@***.***> </div><div>Date: 06/03/2022  21:56  (GMT+00:00) </div><div>To: Ultimaker/Cura ***@***.***> </div><div>Cc: Chas Robertson ***@***.***>, Chas Robertson ***@***.***>, Author ***@***.***> </div><div>Subject: Re: [Ultimaker/Cura] Unable to change infill density with support blocker (Issue #11617) </div><div>
</div>

>>> "Mark Burton" ***@***.***> 03/06/2022 21:57 >>>
What happens if you use different infill patterns in the 100% region, i.e. lines or zig-zag?

smartavionics, 2022-03-08:
> Any ideas?

Sounds like there's some under-extrusion, try printing slower and maybe check the extruder is calibrated correctly?

chasrobertson, 2022-03-08:
If Sent from my Galaxy<div>
</div><div>
</div><div>-------- Original message --------</div><div>From: Mark Burton ***@***.***> </div><div>Date: 08/03/2022  09:07  (GMT+00:00) </div><div>To: Ultimaker/Cura ***@***.***> </div><div>Cc: Chas Robertson ***@***.***>, Chas Robertson ***@***.***>, Author ***@***.***> </div><div>Subject: Re: [Ultimaker/Cura] Unable to change infill density with support blocker (Issue #11617) </div><div>
</div>

>>> "Mark Burton" ***@***.***> 03/08/2022 09:08 >>>
> Any ideas?

Sounds like there's some under-extrusion, try printing slower and maybe check the extruder is calibrated correctly?

fvrmr, 2022-03-15:
Hi @chasrobertson thank you for your report.
Is it working as expected?
Could you maybe share your project file? (file - save project) And share as a .zip here on Github, reply via e-mail will not attach the project file.

closed 2022-03-29 by no-response[bot]