Ultimaker/Cura #11618: Update profile with current settings/overrides. Greyed out and won't work

oscarmemo, 2022-03-07:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

CR-10

### Reproduction steps

Quitting Cura and trying again.
Restarting computer
swap to ender 3 pro profile and back (CR-10 can update profile)
checked all plugins were enabled
Updated Cura to 4.13
changed between profiles while keeping settings and not keeping
[image]

### Actual results

No change button is still greyed out and won't perform action

### Expected results

Current settings should be saved to profile

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

64bittuning, 2022-03-09:
Have you tried actually changing any settings before trying to save them? I build profiles like this all the time.

nallath, 2022-03-09:
Well, I don't see a star in the dropdown, so I'm guessing that no settings have been adjusted and as such, all options are greyed out.

closed 2022-03-24 by no-response[bot]