Ultimaker/Cura #11628: Printer Setup (Size, # extruders, etc) missing in 4.13.1

inspectionsbybob, 2022-03-07:
### Application Version

4.13.1

### Platform

Win10

### Printer

Custom JGMaker Artist-D

### Reproduction steps

4.13.1 installed itself and all my profiles and printers disappeared.

Went to add a printer, but the add section no longer asks or has the option for setting printer options such as Size, Shape, number of extruders, heated bed, and more....

How do I re-create my JGMaker Artist-D Printer??

### Actual results

no way to add my printer

### Expected results

Entire add printer sections are missing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NO more info

fvrmr, 2022-03-10:
Hi @inspectionsbybob thank you for your report.
When installing a new version of Cura you will get a pop up with the question to remove your old configuration files. See https://github.com/Ultimaker/Cura/issues/8233

To add your printer you can use the Custom FFF printer option in the add machine menu. Let me know if that works for you.

Ghostkeeper, 2022-03-20:
Did you create a custom .def.json file? If so, you may want to add the `MachineSettingsAction` to the metadata, like here: https://github.com/Ultimaker/Cura/blob/40fc587d640dd336330e96b6aa9041f1d945ae02/resources/definitions/custom.def.json#L22

If you created that printer simply as a Custom FFF Printer from the interface, the machine settings should pop-up during creation of the printer and should also still be accessible through the printer list in the preferences screen.

inspectionsbybob, 2022-03-20:
something with the upgrade corrupted something and the option for adding
stuff wasnt there.  I cleaned up the appdata files and re-installed and
everything was happy.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2022-03-21 by fvrmr

inspectionsbybob, 2022-10-11:
I had to do a "clean" install. Once I removed the old user data files the
menu to add a printer showed up again and I was able  to proceed....

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016