Ultimaker/Cura #11629: Text overflow on monitor screen

SanduRajapakse, 2022-03-07:
### Application Version

4.13.1

### Platform

MacOS 12.2.1

### Printer

Ender 3 Pro

### Reproduction steps

Print something with OctoPrint. Use smaller window size (not full screens size). View Monitor.

### Actual results

Content overflows past end and behind progress bar.
[image]

### Expected results

Not overflow, maybe allow scrolling for overflow content.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

UI Issue. Screenshots provided.

fvrmr, 2022-03-11:
Hi @SanduRajapakse thank you for your report.
For the next release we have updated all Qt controls which includes this page. I will check on Monday if it fixed with the update. 
Keep you posted.

fvrmr, 2022-03-17:
The update for this page is not finished yet, when it is done. I will update you.

SanduRajapakse, 2022-03-21:
I appreciate it. Thanks!