Ultimaker/Cura #11630: Cura 12.1 uses unused tool to write purge tower

Anathema-Device, 2022-03-08:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Geeetech A10M, Klipper, ERCF

### Reproduction steps

Defined a printer with 8 extruders that share a common nozzle
Enabled first 4 extruders (0-3)
disabled last 4 extruders (4-7)
instantiated the 2 colour test cube with T0 for skirt, T1 & T3 for the cube colours
Specified a purge tower to clean out the old colour
Sliced and sent to Klipper

### Actual results

Edit the resultant gcode in Fluidd to see that there are segments set to T4 that appear to be for the prime tower.  The sliced file is not purging the last nozzle just switching to T4 (that is disabled in Cura and not loaded with filament on the printer)

### Expected results

The current extruder should be run out to the purge tower and filament run through it to push out the old colour before starting with the new colour

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I will attach the project file when I can figure out how to do it

Anathema-Device, 2022-03-08:
[Cura_Bad_T4.zip](https://github.com/Ultimaker/Cura/files/8201986/Cura_Bad_T4.zip)

Anathema-Device, 2022-03-08:
I find that this also happens with Cura 4.13.1

fvrmr, 2022-03-15:
Hi @Anathema-Device thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8213 
I will close this issue since it is a duplicate.