Ultimaker/Cura #11631: Crash when slicing a file

dsvi, 2022-03-08:
### Application Version

4.13.1

### Platform

Linux

### Printer

E3

### Reproduction steps

loaded model, multiplied it by 39, sliced.

### Actual results

Experienced a crash
Tried Cura 4.12, it also crashes 

### Expected results

To not crash

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[crash.3mf.zip](https://github.com/Ultimaker/Cura/files/8207807/crash.3mf.zip)
Had to add .zip extension since Github does not allow to upload .3mf
The file is not actually a zip, it's 3mf

smartavionics, 2022-03-08:
It's a bug which looks like it's been there quite a few years, amazing it hasn't been spotted before.

Anyway, I fixed it and now get this...

[image]

Cura devs, here's a patch from my build, it may apply to master but as I can't build the UM master any more, I don't know.

```
diff --git a/src/FffGcodeWriter.cpp b/src/FffGcodeWriter.cpp
index 3284d1424..1240d4fff 100644
--- a/src/FffGcodeWriter.cpp
+++ b/src/FffGcodeWriter.cpp
@@ -943,12 +943,13 @@ LayerPlan& FffGcodeWriter::processLayer(const SliceDataStorage& storage, LayerIn
     coord_t max_inner_wall_width = 0;
     for (const SliceMeshStorage& mesh : storage.meshes)
     {
-        max_inner_wall_width = std::max(max_inner_wall_width, mesh.settings.get<coord_t>((mesh.settings.get<size_t>("wall_line_count") > 1) ? "wall_line_width_x" : "wall_line_width_0"));
+        coord_t inner_wall_width = mesh.settings.get<coord_t>((mesh.settings.get<size_t>("wall_line_count") > 1) ? "wall_line_width_x" : "wall_line_width_0");
         if (layer_nr == 0)
         {
             const ExtruderTrain& train = mesh.settings.get<ExtruderTrain&>((mesh.settings.get<size_t>("wall_line_count") > 1) ? "wall_0_extruder_nr" : "wall_x_extruder_nr");
-            max_inner_wall_width *= train.settings.get<Ratio>("initial_layer_line_width_factor");
+            inner_wall_width *= train.settings.get<Ratio>("initial_layer_line_width_factor");
         }
+        max_inner_wall_width = std::max(max_inner_wall_width, inner_wall_width);
     }
     const coord_t comb_offset_from_outlines = max_inner_wall_width * 2;

```

smartavionics, 2022-03-08:
Of course, the bug only occurred when initial layer line width was > 100% and multiple meshes were present.

fvrmr, 2022-03-10:
Hi @dsvi thank you for your report.
And @smartavionics thank you for your patch, I will bring it up with the team.

rburema, 2022-08-18:
Finally working on the ticket. Seems we already fixed it at some point, see this commit that is very similar to the proposed patch: https://github.com/Ultimaker/CuraEngine/commit/a0b059c9a3e05c5052d2e9078b2ac8f54a2f4fef

Tested it on both 4.12 and 5.1, the former indeed crashes, but 5.1 is well beyond the commit and slices without crashing.

Thanks for the analysis in any case.

closed 2022-08-18 by rburema