Ultimaker/Cura #11632: Skirt Issue

Halerod, 2022-03-09:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender 3 & Elegoo Neptune 2

### Reproduction steps

Skirt only goes around portion of mode
[image]
l

### Actual results

Don't know

### Expected results

Skirt should go around entire model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

New, need help
[cura.log](https://github.com/Ultimaker/Cura/files/8212109/cura.log)

GregValiant, 2022-03-09:
A project file ("File / Save Project") would be better.  From here it looks like only that one corner is actually touching the build plate.  You can try to use the Rotate Tool "Lay Flat" command or move the model down into the build plate by .1mm (or something).  You may have to turn off "Automatically drop models to build plate" in order to do that.

Halerod, 2022-03-11:
Greg,

 

You were correct, not all of the model was on the build plate.

The lay flat tool did not work and I could not find where to drop the model down by .1mm nor could I find the "Automatically drop models to build plate" in the settings either.

But I was able to rebuild the model and make all the parts even in size before opening in Cura. I also initially tried to save the project and attach a project file. 

When I tried to attach or drop it in the report it said it did not support this type of file. Not sure what I was doing wrong. I ‘d love to know where the settings are you 

referred to and what I was doing wrong when saving and attaching a project file. I know I will have other issues in the future and would like to properly submit issues.

Thank you very much for your help.

GregValiant, 2022-03-12:
I think you can post text files here.  Everything else needs to be Zipped in order to post.

"Automatically drop models to build plate" is in the Menu command "Preferences | Configure Cura | General | Viewport Behavior.  When the setting is turned off you can move the model in the Z using either the grip arrow and dragging it, or just typing in a Z value in the Move Tool box.

The Lay Flat tool only works when the bottom surface of a model is planar.  Fixing the problem in the CAD program was the right way to do it.

closed 2022-03-29 by no-response[bot]