Ultimaker/Cura #11636: Slicing Bug

BirdHunter007, 2022-03-09:
### Application Version

4.13.1

### Platform

Windows 10 Enterprise

### Printer

Ender 5 Pro

### Reproduction steps

Adjusted infill density and infill pattern

### Actual results

Reports a bug while slicing. Program will not slice model anymore.

### Expected results

A sliced model
[CE5_PETG_Form Tip Shaved_Left Half_Scaled_Threaded.zip](https://github.com/Ultimaker/Cura/files/8218080/CE5_PETG_Form.Tip.Shaved_Left.Half_Scaled_Threaded.zip)
.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I was able to slice this project earlier in the day. Then all of a sudden it started to report a bug at the beginning of the slicing process.

GregValiant, 2022-03-10:
It sliced for me in 4.13.1.  I checked using the Mesh Tools and the big parts aren't watertight.  That can confuse Cura.

Here is the report from [https://formware.co/OnlineStlRepair](https://formware.co/OnlineStlRepair)
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 3 Non-manifold edges (?)
--> 5 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 135948ms ------
-> Vertex count changed from 192609 to 195723 (+3114)
-> Triangle count changed from 385224 to 392048 (+6824)

fvrmr, 2022-03-11:
Hi @BirdHunter007 thank you for your report.
I could slice your project with no problems.
[image]
Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

closed 2022-03-25 by no-response[bot]