Ultimaker/Cura #11638: Some walls are considered 'thin walls' when they shouldn't be

hoffr, 2022-03-10:
### Application Version

4.13.1-amd64

### Platform

Windows 10 64-bit, RX570 GPU, all parts and OS tested stable this month

### Printer

Heavily modded Ender 3

### Reproduction steps

1. Reinstall Cura, ensure no forks are installed, manually delete config folders, relevant registry entries, other folders
2. Load default profile for Ender 3, toggle setting visibility to all
3. Set nozzle size to 0.6mm
4. Put my .stl on the plate, set all line width to 0.72mm, wall count to 1, enable 'Print thin walls'
5. Slice and observe that the mesh with the exact width as the line width is considered a thin wall, and will not print without 'Print thin walls' enabled

### Actual results

A 0.720000mm thin rectangular prism is slicing as a 'thin wall' (smaller than nozzle size), despite all line widths being set to 0.72mm.

### Expected results

A 0.720000mm thin rectangular prism should have its wall sliced with a single 0.72 extrusion if 0.72 line width is specified in the profile, and should not be considered a 'thin wall'.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Cura reports accurate scale info for the model on the plate. Increasing the size of the mesh (eg. from 0.72mm to 0.73) causes the wall to print not as a thin wall, but adds an extra undesirable flow-reduced extrusion path overlapping it.

[project_and_log.zip](https://we.tl/t-jvSYHXr3gN)

Asterchades, 2022-03-10:
It's been a long-standing issue that walls exactly as wide as the line width can sometimes be cropped when they shouldn't be. Increasing the object width by 1um or lowering the line width by 1um will work around it.

Cura 5 shouldn't have the problem, though. Arachne is far more flexible than the current system, especially when it comes to walls that are exact multiples of line widths.

hoffr, 2022-03-10:
Wall line width had to be reduced from 0.72mm to 0.695mm for all the walls to slice non-thin. Only the wall facing the other axis vs the other two sliced non thin above 0.695mm, all the way up to 0.717mm.

Arachne Beta 2 is interesting but it doesn't exactly fix the issue, it just widens the line width and there's no way to disable that behaviour.

I appreciate the tips though. I'm trying to work with very precise tolerances and this is sketching me out

fvrmr, 2022-03-11:
Hi @hoffr thank you for your report.
@Asterchades is correct. In Arachne this should work better with this. 
Did Arachne work for you? If not could you show me why not.

hoffr, 2022-03-11:
@wolfgangmauer I think it does solve the problem I'm having, but I'm currently married to smartavionics' Cura so I'm not sure it'll always be useful to me! lol. I am also not really one to use a beta, but I may as well give it a try. Thanks, this can probably be closed. Seems like some great work is being done on the next engine.

fvrmr, 2022-03-14:
Thanks for the reply. Closing this issue then.

closed 2022-03-14 by fvrmr