Ultimaker/Cura #11644: When using brim with the option only on the outside it does not print the firt

Unufry, 2022-03-10:
### Application Version

4.13

### Platform

Windows 10

### Printer

Two trees sp-3

### Reproduction steps

When using brim with the option only on the outside it does not print the first layer of support material!
Upgraded to the latest version, the problem is still ocurring.
Added 3 pictures for visual asistance.

### Actual results

When trying to print it skips the fist layer and the support structure does not stick to the build surface.

### Expected results

Expecting when the option brim only on the outside is activated to include support material where the brim is supposed the be.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]

GregValiant, 2022-03-10:
You may have to enable "Enable Support Brim" in the Support section.  (Those lines should stick though.)

Ghostkeeper, 2022-03-20:
Seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/9996?

closed 2022-03-25 by no-response[bot]