Ultimaker/Cura #11645: Using support blocker with single object + One at a Time print sequence results in reduced print volume height

ansonl, 2022-03-10:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Ultimaker Original+

### Reproduction steps

1. Add one object to build volume that is taller than gantry height
2. Set Print Sequence to One at a time. 
3. Add support blocker

### Actual results

The support blocker is treated as a second object and the build volume height is reduced below the imported object height and slicing fails. 

### Expected results

The build volume height should not be reduced to gantry height (below the imported object height) after support blocker is added to a single object print with "One at a time" sequence because there is just one object. Slicing should work just like when there is a single tall object being sliced with sequence set to "one at a time" to avoid confusion.  

Even though slicing a single object is the same with All at Once and One at a Time sequencing can can be fixed by changing the sequence, it should not fail with a single object in One at a Time mode due to supposedly treating support blocker as a second object. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Add support blockers for the holes to test. 
[fanduct.zip](https://github.com/Ultimaker/Cura/files/8226917/fanduct.zip)

fvrmr, 2022-03-11:
Hi @ansonl thank you for your report.
I will bring it up for discussion. Keep you posted.

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If the problem persists into Cura 5.13.0 please fill out a new report so it goes to the front of the queue.

closed 2026-09-07 by GregValiant