Ultimaker/Cura #11646: Support Interface Unusual Behaviour

3Duke3, 2022-03-11:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Ultimaker 3 ext

### Reproduction steps

Import model with large flat surface containing holes that needs support material. Scroll to layer previews of support interface, layer 112-118

From left to right.
Left - Model as required
Center - Model as above but with .2mm sacrificial surface covering holes
Right - Model as above but with no holes at all for comparison.

### Actual results

With the model on the left, the support interface layers are printed with nozzle 2, however nozzle 1 is printing small rings that line up with the model holes inside every support interface layer. This creates some problems as these small printed circles/rings seem to get pulled off the support interface and dragged through the print. (I don't remember this behavior in previous versions of Cura and cant see why they are now being generated)

With the model on the far right there are no holes and the support material and interface is generated as we would expect.

However, with the center model, I have included a sacrificial .2mm layer that covers the holes. You would expect the support interface behavior to be identical to the model on the right. However, Cura is generating the same small rings that are being printed with extruder 1, expect now they are alternating throughout every second interface layer. This leads me to believe there could be a bug?

In the left and center model, the last support interface layer is less than ideal for printing on top of. The right model's top support interface layer is perfect for printing on top of. If the right models support interface could be the generated result for all these models, this would be beneficial, ie support interface printed as a solid flat regardless of having a model hole above it that is less than say 10mm.

[image]
[image]

### Expected results

Support interface generated with exact profile of surface being supported.

Feature request: If the supported surface has small holes under a set diameter, just print a complete flat interface without holes.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Support Interface Example.zip](https://github.com/Ultimaker/Cura/files/8228951/Support.Interface.Example.zip)

GregValiant, 2022-03-11:
It is odd.  The circles only show up every other layer.  When I switch printers to my E3Pro the condition goes away.

3Duke3, 2022-03-12:
It appears adjusting the support interface resolution setting impacts this behavior.

GregValiant, 2022-03-12:
It may be a setting but I don't see it either.
This model has holes of various sizes.  Starting with the 4mm hole each has walls from the floor to the start of the interface.
[image]

But the 2, and 3mm holes only appear in the interface and they would just drop through to the floor since there often would be no support below them.  The 1mm hole must be below some threshold as it is ignored.
In regards to the  2 and 3mm holes - those supports are much smaller than the holes.  This appears to be similar to what @3Duke3 has when I slice his original posted file.  Those circles in the interface don't match the inside diameter of the holes.
[image]

fvrmr, 2022-03-14:
Hi @3Duke3 thank you for your report. 
I will bring it up for discussion, keep you posted.

3Duke3, 2022-03-15:
@fvrmr I think the bridging of support interface over small holes would be a feature request, but the main issue is really the generation of the circles with extr1 inside the support interface layer of extr2. 

As @GregValiant pointed out, it seems to be related to holes in the model being 3mm or under (perhaps there is a correlation to support density? I will check)

but the fact these circles appear on every second layer lead me to believe there is an anomaly here. The everysecond layer outcome can be influenced by the support interface resolution, but you can only make the circles appear on every layer rather than removing them completely which is really what needs to be happening.

Printing such small circles at speed, on the top of breakaway means the circles do not adhere well to the surface. They are often dragged out of place by extr1 or dislodged by extr2 as it dances around them while making the next layer of interface. 

I don’t remember seeing this behaviour on some earlier versions of Cura (would be at least a few version ago) but I also can’t remember if I have had to make such small pilot holes on top of a support interface before.

Ghostkeeper, 2022-04-02:
I think actually that this is simply the result of the Support Interface Resolution setting. What happens if you set that to 0.15mm (1 layer height)? For me that seems to resolve the issue.

Looking at the description of Support Interface Resolution, that seems to be expected behaviour. It's a limitation where fixing it would make slicing slower for everyone.