Ultimaker/Cura #11648: Slicing generates a vertical artifact in cylinder based model

steveb48, 2022-03-11:
### Application Version

4.13.1

### Platform

macOS 12.2.1

### Printer

Ender 5 Plus

### Reproduction steps

Import model from Fusion 360
Select Ender 5 Plus (same problem if you select Ender 5 Pro)
Slice
Switch to Preview

### Actual results

View the edge of the model and you'll see a vertical line
I wasn't able to find any setting which would change it.
This appears when printed as well.
[image]
[Cylinder artifact problem.3mf.zip](https://github.com/Ultimaker/Cura/files/8234354/Cylinder.artifact.problem.3mf.zip)

### Expected results

No artifacts 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Not sure what's required here. Project and Screenshot were uploaded above.

64bittuning, 2022-03-11:
That is the seam. Cura isnt doing anything wrong.

steveb48, 2022-03-11:
Thanks for the speedy reply.

What seam are you referring to? If I model a cylinder in Fusion 360 (just an extruded circle), import it into Cura 
I see this “seam”. Simplify3d doesn’t appear to have the same issue although I’ve only viewed it on screen, not printed it.

Steve

64bittuning, 2022-03-11:
Each layer has to start and end somewhere. 

[image]

64bittuning, 2022-03-11:
Every layer has to start and stop somewhere...

https://www.youtube.com/watch?v=NU1kYEE3qrQ

steveb48, 2022-03-11:
Great video! Now I understand. 

thanks,
Steve

fvrmr, 2022-03-14:
Closing this issue since this is not a bug. 
Thank you @64bittuning for responding.

closed 2022-03-14 by fvrmr