Ultimaker/Cura #11649: Cura crashing after slice

bflyth, 2022-03-11:
### Application Version

4.13.1

### Platform

Windows 10.0.19043 Build 19043; RTX 2080; Ryzen 5 OC 4GHz

### Printer

Ender 3 V2

### Reproduction steps

1. Open any stl or object file
2. adjust settings to slice depending on object needs (rafts, supports, etc)
3. slice
A4. inspect object layer by layer
or
B4. leave software open after slice

### Actual results

A5. crash
B5. Crash

### Expected results

A5. layers render every time with no problem

B5. software stays open with no problem

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/8236342/stderr.log)

bflyth, 2022-03-11:
This time the application crashed after about half an hour of idle time

fvrmr, 2022-03-15:
Hi @bflyth thank you for your report.
Do you have an external monitor?
Or it could be related to https://github.com/Ultimaker/Cura/issues/6855

bflyth, 2022-03-17:
> Hi @bflyth thank you for your report. Do you have an external monitor? Or it could be related to #6855 could you maybe check if this helps for you: [#6855 (comment)](https://github.com/Ultimaker/Cura/issues/6855#issuecomment-650519439).

Well, yes I use an external monitor; this is a desktop PC. I will attempt WIN8 Compatibility and let you know. 

This crashing fiasco is really making the software undesirable. I'm only sticking around for Gyroid infill.

closed 2022-03-31 by no-response[bot]

bflyth, 2022-03-31:
Re open this. I answered your question. The problem persists. Cura has crashed THREE times in the past hour while I'm trying to work.