Ultimaker/Cura #11651: Cura lagging and not responding

gncobb, 2022-03-12:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

I have been troubleshooting this for weeks.  This didn't begin until 4.13 and not instantly.   I have re-installed, did computer updates, deleted old Cura folders, removed plug-ins from Cura, have disabled virus scanning on the Cura folders, and have done more than the normal user could imagine knowing how to do.  This is not specific to a certain size STL file. 

### Actual results

When I open the program it lags for about 4 seconds.  After that I can move the table.  I can view and change the profile settings.  If I load a STL it will almost always take around 45 seconds to show on the table.  If I click as soon as it loads, it will lag for 30-45 seconds.  If I click while it is lagging the title bar will show Not Responding for a minimum of one minute.  The more I click the longer it lags.  This is also true with the Configure option, Market, menubar on the left, when I am in preview mode moving around, etc.  It is slow slicing.  Any time I move around fast it will hesitate, so I have to wait a few seconds to give it breathing room before clicking.  I was trying to close out of the Settings dialog box earlier and it lagged for so long the program crashed.  -  I didn't capture but part of this from today, will try to find the other file if I hadn't deleted it.   -   I also checked Task Manager and it showed 1.1GB of memory being used by a single instance of Cura.

### Expected results

I have a new I7 with SSDs.  I shouldn't be lagging at all.  I just need something I can use.  If you need me to do more testing or install a beta copy at some point let me know.    gcobb@photogesture.com

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8237050/cura.log)
[cura-4.13.1-22-03-11-18_45_54.zip](https://github.com/Ultimaker/Cura/files/8237052/cura-4.13.1-22-03-11-18_45_54.zip)
[cura-4.13.1-22-03-11-16_55_26.zip](https://github.com/Ultimaker/Cura/files/8237053/cura-4.13.1-22-03-11-16_55_26.zip)

molekm, 2022-03-13:
I have similar issues, i9 9900k , 64gb ddr4, nvme's .
CUra 4.13 is deadly slow to respond on any action in Materials, it takes 5-10seconds to click / focus on a textbox.

fvrmr, 2022-03-15:
Hi @gncobb thank you for your report. 
I will share your files from the plugin with one of our developers for analysis. Keep you posted.

gncobb, 2022-04-08:
Well...It rarely works as it should still.

Ghostkeeper, 2022-04-10:
Here is a part of the results of that profiling.
[image]

The time is spent by `_GenericAlias.__hash__`, a function inside of the `typing` module. It's doing that because of `ShaderProgram._setUniformValueDirect`, a function that's being called ~12 times per render to send parameters to the GPU. If this is causing the slowdown, you'd be looking at low framerates, not at a 4-second freeze during start-up. It looks like this is just significant in the result because it renders very often.

Another significant delay is `LayerPolygon.__init__`, which is where it processes the layers for layer view after completing a slice. This takes about 2 seconds which is about what I'd expect for a normal slice. There is a progress bar there, although it is a bit choppy due to this function taking up Python's GIL.

The rest of the functions here are either part of rendering or starting to get too insignificant to be causing a 4-second delay. So I would conclude that the delay is caused by something outside of Python's function calls, e.g. by Qt during the redraw of the interface. One thing that might help is to reduce the number of visible settings to just what you need, rather than showing all of the settings all the time.

We might be looking at another case of https://github.com/Ultimaker/Cura/issues/10936 here, although the profiling is a bit different. The symptoms are similar.