Ultimaker/Cura #11652: "There was an error while uploading your backup"

GoguyT3D, 2022-03-12:
### Application Version

4.13.1

### Platform

Windows 11, Gigabyte Geforce GTX 1050 Ti 4GB GDDR5 128 Bit PCI-E Graphic Card (GV-N105TD5-4GD)

### Printer

Ender 3 Pro

### Reproduction steps

Opened cura and "Please wait while we upload your backup" was displayed
Then "There was an error while uploading your backup"

### Actual results

Some settings seem revert back to older setting and not updated to my most reecent changes

### Expected results

My backup should be loaded and my settings should be loaded as they were from the last time I closed cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8237698/cura.log)

fvrmr, 2022-03-14:
Hi @GoguyT3D thank you for your report.
In your log I can see that you are not logged in. Which is weird since you need to be logged in for creating back up. Could you try to log out and log in again and try the back up again.

GoguyT3D, 2022-03-19:
Yes the backup works when I am logged in, although when I start cura fresh for the first time in a day I am not automatically logged quickly enough and it fails to upload the backup automatically.

fvrmr, 2022-03-23:
Thanks, I will bring it up with the team. 
Keep you posted.

fvrmr, 2022-03-24:
Do you get an error as well when you want to open the marketplace?

GoguyT3D, 2022-03-25:
No marketplace opens and loads fine

GoguyT3D, 2022-07-10:
Even on 5.0 and 5.1 beta the application loads so quickly and attempts to upload my backup before the app even logs me in and it fails. But sometimes if the app has been closed and is reopened during the same PC power cycle it doesn't give me the error.

klepp0906, 2022-07-28:
just found this as i'm experiencing the same thing

jacobhanson1010, 2024-01-16:
I'm having this same problem. Each day that I start cura, it fails to create a backup.

Shootingblanks00, 2024-07-18:
same thing happening to me. 5.7.1