Ultimaker/Cura #11654: Disable z hop in top layers

Mcklp, 2022-03-12:
### Is your feature request related to a problem?

Zhopping in top layers sometime leaves some filament in place where its is zhopping. Its not a problem in walls and inside model, but on top its anoying.

### Describe the solution you'd like

Check box in option that will turn off z hop in all top layers(or in last vissible one if its possible). 

### Describe alternatives you've considered

-

### Affected users and/or printers

It should make top of flat things nicer, while not turning off z hop.

fvrmr, 2022-03-15:
Hi @Mcklp thank you for your feature request.
Seems like this is related to https://github.com/Ultimaker/Cura/issues/10145 but there it is only for the first layers.
I will bring this up for discussion with the team. Keep you posted.

fvrmr, 2022-03-21:
We have discussed it with the team and we decided to defer this request. It has not a high priority to implement this. I hope you understand.

karcaw, 2023-06-29:
I would really like this in cura.  It would greatly help me.  It is the biggest reason I run a prusa slicer, since they have this feature, and for a few of my specific multi color prints i need to turn off zhop after the multi-colors go down in the first .6 mm, after that i dont need zhop at all.  if this was in cura, i wouldn't need to switch to prusa to do those prints.

krokus57, 2023-06-30:
Hi! This is very necessary for delta printers. It will allow you not to touch the part with the nozzle and increase the printing speed while reducing the number of retracts.