Ultimaker/Cura #11655: Bottom layer gaps/incomplete

naasking, 2022-03-12:
### Application Version

4.13.1

### Platform

Windows 10 64-bit

### Printer

Kingroon KP3 with Marlin firmware

### Reproduction steps

Printed the model titled "Drawer_1x2x1_clean.stl" from [this page](https://www.thingiverse.com/thing:3873672/files). 

### Actual results

The final result was still perfectly usable, but the printer consistently missed a portion of the handle on one of the bottom layers leaving an artifact you can see here:

https://imgur.com/a/pOTnKJ8

### Expected results

The bottom layers should be completely filled.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The attached project file has another object in it, but I've printed the drawer on its own as well and the exact same gap was there.

GregValiant, 2022-03-13:
Use the "File | Save Project" command to create a 3mf project file.  Zip it up and post it here.

naasking, 2022-03-13:
Hmm, I did attach it, not sure why it didn't stick. Here it is again.

[PLA-box connector+drawer.zip](https://github.com/Ultimaker/Cura/files/8239564/PLA-box.connector%2Bdrawer.zip)

GregValiant, 2022-03-13:
I tried but I can't duplicate the problem.
The Cura gcode reader is separate from the slicer.  If you open that Gcode file in Cura and look at the second layer - are those 5 missing lines in it?    (I've never seen Cura miss lines like that before.)

naasking, 2022-03-13:
No, everything looks fine, just like the image I posted above of the preview. I haven't had any missed lines in the couple of years I've used Cura either. I suppose it could be a Marlin bug.

Maybe I'll try slicing the same model with SuperSlicer and see if it happens with that too. I'm printing a bunch of these so I'll have ample chances to try different things if you have any suggestions.

GregValiant, 2022-03-13:
If you are printing from the SD then another failure option could be a bad sector on the card.

fvrmr, 2022-03-14:
Hi @naasking thank you for your report. 
I can't reproduce your issue in Cura. So I suppose it is related like @GregValiant mentioned your SD card. Does it just skip these lines and move to the next layer?

naasking, 2022-03-22:
I tried a low-level format of the card and the issue remains, and the issue has since turned up in other printed models of different sizes. I might try another SD card at some point, but I think this is a firmware issue, so no need to bother with this any further as it's a fairly minor issue that has no impact since it's only ever on the bottom layers.

closed 2022-03-22 by naasking