Ultimaker/Cura #11656: Super wrong scaled GUI under Gnome

zewonex, 2022-03-13:
### Application Version

4.11.0

### Platform

Linux Fedora; Gnome under Wayland; Cura as .rpm

### Printer

Doesn't matter

### Reproduction steps

1. open Cura
2. open file and slice model
3. see, that you can't do that because UI is bugged
4. switch to fullscreen, now everything is fine
5. put scaling back to 100%, not 200%; now everything is extremely small, but "normal"
6. switch to xorg instead of wayland. same problem
7. switch to KDE Plasma (same distro, other desktop). Now, every scaling option works. 100, 150, 175 and 200%. 
8. go back to gnome, because gnome is bae
9. rescale the cura window from whole-screen (not fullscreen) to partial screen. won't go smaller

### Actual results

see above

### Expected results

proper scaling not really working with gnome

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]

fvrmr, 2022-03-14:
Hi @zewonex thank you for your report.
Does this happen with the AppImage? You can find the AppImage here: https://github.com/Ultimaker/Cura/releases

zewonex, 2022-03-14:
It happens only on the .rpm-version.
The flatpak-one and the appimage are both just as buggy as hell and unusable as every version has its issue on its own.
The scaling-bug on the rpm happened after an update of the app.

fvrmr, 2022-03-16:
We don't maintain that version, only the AppImage. Report your issue to the maintainer of that package.
So I will close this issue, if you have issues with the AppImage please make a new bug report.