Ultimaker/Cura #11658: Make the Overhang Shader an actual shade not a solid colour

3Duke3, 2022-03-13:
### Is your feature request related to a problem?

When preparing models with technical overhang geometry, the red overhang shader is a solid colour that blocks visible on screen inspection of the actual underlying part geometry. This makes it hard to visually inspect overhangs and place/manipulate support blockers in their appropriate locations

### Describe the solution you'd like

I think it would be ideal if the red overhang shader was in fact a shade of red rather than the solid colour red, something that let the highlights and shadows of the model bleed through so we could still see the underlying model geometry. All other visual aspects of a 3d model are shaded with highlights as described, making it easy to see the geometry of a model.

[image]

When the overhang shader is applied to the surface, the models highlights and shadows disappear making it impossible to see the geometry of the model.

[image]

The pictured images are a crude example but convey the point of detail hidden by the overhang shader. Sometime the geometry is more pronounced and it makes placing support blockers a guessing game.

[image]

When visually inspecting a model the overhang shaders red colour is already extreemly noticeable, I feel even a subtle ghosting of red over the overhang surface would still be very noticeable and just as effective, with the added benefit of being able to see the geometry.

### Describe alternatives you've considered

Set overhangs angle value to 90 to inspect model without any overhang shader, then change to desired angle for printing. 

### Affected users and/or printers

Everyone.

fieldOfView, 2022-03-14:
The whole idea is to set the overhang shade apart from other parts. If it were shaded, it would be entirely impossible to distinguish the overhang indication from a red model (when printing in a red material). 

You can turn off the overhang visualization in the General preferences. It could be nice if the prepare view had a legend like preview view, where there could be a checkbox to show/hide the overhang visualization like it is possible to show/hide helpers and travel in the preview.

Having said that, I think there's a lot to be said about changing how the overhang is visualized. The "flat" red color suggests an error, and in other parts of Cura a red error blocks slicing. Some rethinking of all the different colors and patterns in the prepare view could be a very good thing.

3Duke3, 2022-03-14:
I understand the red is to set it apart from other colours to indicate overhangs, I just mean it would be good if the red still showed the underlying detail detail of the model by use of shadows/highlights. 

When looking at the current red, because its a flat solid colour, any surface it covers now looks 2d to the eye... hopefully that makes sense?

Excuse my poor photoshop skills, but something along the lines of looking like the below is what I had in mind for highlighting overhanging surfaces. The overhang surface is still highlighted red, but because there are shadows/highlights now in the red, I can see exactly where I am trying to place a support blocker.

[image]

fvrmr, 2022-03-15:
Hi @3Duke3 thank you for your feature request.
I will bring this up with the team. Keep you posted.

3Duke3, 2022-03-15:
Thanks @fvrmr I think the biggest selling point is just that it would make for a more refined/professional/modern looking UI for a better user experience. Not a necessity as the current works, just visually more refined.

fvrmr, 2022-03-21:
We created a ticket for this on our backlog. Devs see CURA-9057