Ultimaker/Cura #11661: AppImage is now missing GLIBC 2.28 in 4.13.1 on Linux Mint 19.3

Area5142, 2022-03-14:
### Application Version

4.13.1

### Platform

Linux Mint 19.3 x64

### Printer

Any

### Reproduction steps

Trying to execute **_Ultimaker_Cura-4.13.1.AppImage_** from command line.

### Actual results

Cura aborts with:
cura: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by cura)

### Expected results

Application would start as in earlier versions. Confirmed version that worked is 4.12.1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None.

fvrmr, 2022-03-16:
Hi @Area5142 thank you for your report.
This is indeed correct, the reason why is because we updated our build servers which has GLIBC 2.28. 
See [this post](https://community.ultimaker.com/topic/39498-cura-4131appimage-version-glibc_228-not-found-required-by-cura/) on our forum, there you can also find a build that should work: https://community.ultimaker.com/topic/39498-cura-4131appimage-version-glibc_228-not-found-required-by-cura/?do=findComment&comment=302543 
We are not planning to update this for next releases. I will close this issue, I hope you understand.

jsjax, 2022-03-26:
This also affects Ubuntu 18.04

eGit, 2022-05-22:
Everything Ubuntu 18.04 (most Linux distris) and below is broken. I could understand an incompatibility update when there is a pressing need. But it seemed the old build server did work just fine. Also 18.04 is basically just yesterday in Linux terms. Even 16.04 is still in heavy use.