Ultimaker/Cura #11669: Cura 4.13.1 unable to slice due to support blockers

Anathema-Device, 2022-03-15:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Geeetech A10M Klipper with MMU

### Reproduction steps

If you enable only one extruder in a multi-extruder configuration then a model will slice fine.  However if you then add support blockers the model will fail to slice as Cura needs extruder 1.  If you then turn on extruder 1 Cura will want to use it to write the support and it is not apparent how to disable this as the support extruder selection is then set to extruder 1 and gray (disabled) so it can't be changed.  The only real work-around is to load extruder 1 with the single colour or use a different slicer.

### Actual results

Impossible to slice without changing the machine filaments

### Expected results

The slicer should just use the enabled filament and slice as it will do without the blockers.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Picture of the message and the project file
[CFFFP_fail_to_slice_due_to_support_blockers.zip](https://github.com/Ultimaker/Cura/files/8256501/CFFFP_fail_to_slice_due_to_support_blockers.zip)
[image]
e

GregValiant, 2022-03-15:
130+ support blockers and they are all assigned to extruder #1.
Support blockers can be scaled like any other mesh.  In the Marketplace you will find a plugin for Cylindrical support blockers.  I'm pretty sure you could get by with two support blockers.  Just bring in one, assign it to extruder #2, and then multiply it.  No work-arounds are required.  You can see the first support blocker at the top.  It is 45mm dia x 20mm tall.  The second is inside the bottom to prevent support on the threads and is 54mm dia. and 15mm tall.  Within the Per Model settings they are set to not support overlaps.  Your other settings require a flow rate of 24mm³/sec?
[image]

[image]

Anathema-Device, 2022-03-16:
Well, that is interesting.  I never imagined that a support blocker was a "thing" that could be sized and assigned to an extruder.  I've just used the tool to drop them in place as they are not something I use much as we mostly create parts that are designed not to require support because the surface is always messy.  So if a blocker is assigned to an extruder why are they generated for extruder 1 when (at the time of creation) only extruder 8 was in use?  And it seemed to force infill to use extruder 1 and be not changable in the settings.

How do you determine the flow rate?  It prints fine at 50mm/s

I only see cylindrical supports in Marketplace, no blocker plugin.

GregValiant, 2022-03-16:
"Custom Support Cylinders" is the one.  You select it and set the Per Model settings to "don't support overlaps".
As far as I can tell, all blockers come in active on Extruder 1.  One of the options for any blocker or Modifier Mesh is "Print as Part" so they have to come in assigned to an extruder and all printers have an Extruder #1.
At a .3 layer height X .8 line width = .24mm².  An extrusion 100mm long is then 24mm³.  Your print speed is listed at 100mm/sec for support and inner-walls and so that is 24mm³/sec.  The outer-walls at 50mm/sec would then flow at 12mm³/sec.

Ghostkeeper, 2022-04-10:
Is this the same problem as https://github.com/Ultimaker/Cura/issues/7428?

Anathema-Device, 2022-04-10:
Looks similar at least, basically the issue is that the support blocker will be assigned the first extruder, not even the first active extruder and the message does not shed any light on the reason.  Part of the issue is that I did not expect a support blocker to be a "thing" assigned to an extruder.  My thought was that it was just a defined area dissociated with any extruder.  Finding it in the list to the lower right and giving it an active extruder makes the slicing possible.

iamVarNat, 2023-09-01:
@Anathema-Device Thank You!