Ultimaker/Cura #11673: Can you add a topmost speed control?

wxtiankong, 2022-03-16:
### Is your feature request related to a problem?

Yes, I print box type things (200 * 200) very slowly, because there is no topmost part speed setting, so I can only set the top and bottom part speeds. I can only compromise this speed. If I can set the speed of the topmost part (lower speed), I will set the bottom and top speed faster, and the overall printing time will be much shorter.

### Describe the solution you'd like

add topmost part speed settings

### Describe alternatives you've considered

no

### Affected users and/or printers

all users

wxtiankong, 2022-03-17:
> With a print speed of 30 and the Search and Replace post-processor active... Search = ;LAYER:396 Replace = ;LAYER:396\nM220 S200
> 
> And now the speed starting at the end of layer 396 will be 200% of 30 = 60mm/sec. (Remember that Gcode layers are Base0).

However, in this way, the inner wall, outer wall and filling speed all will be modified.

wxtiankong, 2022-03-17:
> With a print speed of 30 and the Search and Replace post-processor active... Search = ;LAYER:396 Replace = ;LAYER:396\nM220 S200
> 
> And now the speed starting at the end of layer 396 will be 200% of 30 = 60mm/sec. (Remember that Gcode layers are Base0).

thx

Asterchades, 2022-03-17:
Add one or more Top Surface Skin Layers. These can be almost completely adjusted independently of every other portion - speed, line width, and even pattern. If you add the same number of layers here as your Top Layers they will completely replace them, allowing for you to set the speed of the top and bottom separately.

fvrmr, 2022-03-17:
Hi @wxtiankong thank you for your request.
Like @Asterchades mentioned you can use Top Surface Skin Layers. Here you can read more about this setting: https://github.com/Ghostkeeper/SettingsGuide/blob/b6220f7cc21bac3a470d5632c7be8e72f9ec2012/resources/articles/top_bottom/roofing_layer_count.md
Please let us know if this would work for you.

wxtiankong, 2022-03-18:
> Add one or more Top Surface Skin Layers. These can be almost completely adjusted independently of every other portion - speed, line width, and even pattern. If you add the same number of layers here [...]

thank you very much. I tried, yes.

wxtiankong, 2022-03-18:
> Hi @wxtiankong thank you for your request. Like @Asterchades mentioned you can use Top Surface Skin Layers. Here you can read more about this setting: [...]

thank you very much. I tried, yes.

closed 2022-03-18 by wxtiankong