Ultimaker/Cura #11681: cura crashes when opening files in large icon tiled view & can't change view back to list view

JohnnyMindgames, 2022-03-18:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Accidently clicked on large icon tile view when attempting to load a new file to the build plate. 
2. Cura crashed closing down immediately. 
3. When I reopen settings for file view have been retained and every time I try to open a new file it  crashes before I can change the view to list view. 

### Actual results

[image]

### Expected results

 file opened on build plate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-03-19:
Well crap.  I hope someone comes along with a way to recover from this one.  The Open File dialog no longer works.  I can drag and drop onto the build plate and access recent files but the Open File dialog set to the Large Icon view blows Cura right out of the water.
Here's the log.
[curaOpenCrash.zip](https://github.com/Ultimaker/Cura/files/8308365/curaOpenCrash.zip)
.

smartavionics, 2022-03-19:
Hi @GregValiant , I know nothing about Windows as I don't use it but how about this... if my build isn't affected, use that to open the file dialog and then switch it back to small icons (or whatever works OK) and then close my build and see if UM cura works OK again?

GregValiant, 2022-03-19:
Hi @smartavionics.  I had tried that but I guess the same reason 4.4 through 4.11 and Master were not affected kept your fix from having any effect.
I finally moved all the files out of the Downloads folder.  When I opened it in Cura I was able to change the View back to Details.  After moving the files back into the folder it was OK.  It is odd that simply changing the way the file icons were represented had such a drastic effect.  With the folder empty and no icons displayed I was able to access it to make the change.

I open a lot of other peoples 3mf project files and I had always thought that would jump up to bite me at some point.  It turns out I was looking in the wrong direction.

JohnnyMindgames, 2022-03-28:
Yes, I do have chitubox installed. I ran the fix shown & it worked.  Thank you so much

fvrmr, 2022-03-28:
Thank you for your reply, I will close it this issue.
Duplicate of https://github.com/Ultimaker/Cura/issues/9668

closed 2022-03-28 by fvrmr