Ultimaker/Cura #11683: Cura leave out holes and or tooth in my gears.

DennyNem, 2022-03-18:
### Application Version

15.04 and 3.2.1

### Platform

Windows 10

### Printer

EcubeMaker Fantasy Pro II

### Reproduction steps

Cura leave out holes and or tooth in my gears.
[image]
[image]

### Actual results

the hole in the gear is not there.

### Expected results

there should be a thru hole in the gear
[Gear5.zip](https://github.com/Ultimaker/Cura/files/8306483/Gear5.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8306530/cura.log)

fieldOfView, 2022-03-18:
Neither of those versions is supported anymore. Please try a more recent version of Cura.

Having said that, the model has inconsistent normals, and needs to be "repaired". You can use the Mesh Tools plugin, available from the Marketplace (in Cura 3.5 and newer), to fix the normals, and it will slice correctly after that.

fvrmr, 2022-03-24:
Hi @DennyNem thank you for your report.
@fieldOfView is correct you need to use a newer version of Cura: https://ultimaker.com/software/ultimaker-cura
In 4.13.1 I fixed the model with using the Mesh Tools plugin and then it sliced correctly.
[image]

I will close this issue. But let me know if you have questions.

closed 2022-03-24 by fvrmr

DennyNem, 2022-03-24:
Thank you I will try it.