Ultimaker/Cura #11684: Your configuration seems to be corrupt....

GoguyT3D, 2022-03-19:
### Application Version

4.13.1

### Platform

Windows 11, Gigabyte Geforce GTX 1050 Ti 4GB GDDR5 128 Bit PCI-E Graphic Card (GV-N105TD5-4GD)

### Printer

Ender 3 Pro

### Reproduction steps

Recently I started renaming my printing profiles (the profile names were being renamed to be the same as my second printer) and duplicates of the printing profile were showing up immediately which needed to be removed. After restarting cura I received an error.

### Actual results

After cura has been started I get the error displaying "Your configuration seems to be corrupt. Something seems to be wrong with the following profiles:
- custom_material #14_creality_ender3
- creality_base_extruder_0 #2
Would you like to reset to factory defaults? Reset will remove all your current printers and profiles.

### Expected results

Im guessing because I named my printing profiles the same name as my second printer that the software didn't quite understan
[cura.log](https://github.com/Ultimaker/Cura/files/8309203/cura.log)
[image]
d

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

If I can simply rename my printer profiles and reload them from file (I have exported all of my profiles already) I don't mind "resetting" cura like the application is asking. If all my profiles will be deleted and I need to remake them all from scratch I'd like to try another rout.

GoguyT3D, 2022-03-19:
I just attempted to reload one of my backups from a few days ago and the profile names were "reset" back to the original name before this problem. I did not receive the error "Your configuration seems to be corrupt" although just like my other bug report the backup failed to upload, probably because cura can't log me in fast enough to do the backup. Now I will attempt to rename these profiles again although I will not use identical names for the profiles as I used in my second printer. I will be back shortly with another update.

GoguyT3D, 2022-03-19:
No, I cannot rename my profiles without getting the error "Your configuration seems to be corrupt" How do I properly rename my profiles without this happening? I go into manage profiles, activate the profile I want to rename and click rename and then cura crashes asks if I want to create a bug report, then shows a duplicate of the profile when I reload cura, one with the renamed profile and one with the old name....Now its displaying "Your configuration seems to be corrupt" agian, Im stuck

fvrmr, 2022-03-23:
Hi @GoguyT3D thank you for your report.
Did you rename the file in Cura or did you edit some files yourself in the configuration?

GoguyT3D, 2022-03-23:
I was in cura and viewing profiles and within the app I selected "rename"
and after I renamed it it created a duplicate file with a "2" after it then
claimed my profiles were corrupt.

fvrmr, 2022-03-24:
Thank you that shouldn't happen indeed. I will bring this up with the team. 
Keep you posted.

fvrmr, 2022-04-05:
I have created a ticket on the backlog for this.
Devs see CURA-9085

mrxzenergy, 2022-06-04:
This is easily recreated when you have multiple printers with the same name on a profile as another "incompatible" printer. It will rename the profile with a #2 etc after it even though the "#1" is not on the same printer.

GoguyT3D, 2022-07-10:
Now I have 3 Ender 3 Pro printer profiles in cura and I only own/need 1, maybe in a future upgrade you guys could put a "Join printer profiles" so that all 3 printers share the same related filament and printing profiles.

closed 2022-10-25 by GoguyT3D