Ultimaker/Cura #11685: Wall Thickness {wall_thickness} is looking like a float in GCode

NilsRo, 2022-03-19:
### Application Version

4.13.0

### Platform

Windows 11

### Printer

Anycubic Mega S

### Reproduction steps

Add {wall_thickness} in GCode

### Actual results

Showing 1.2000000000000002
[image]

### Expected results

Should add 1.2 in GCode

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Tageslichtlampe.zip](https://github.com/Ultimaker/Cura/files/8309901/Tageslichtlampe.zip)

GregValiant, 2022-03-20:
What software actually generated that gcode file?  Was it post-processed?  I've never seen all those ";Slicer Info" lines before.

NilsRo, 2022-03-20:
That's included in Printer Startcode to transfer this information to OctoPrint. My plug-in Slicer Estimator shows all metadata there. So no Post-Processing.

fvrmr, 2022-03-21:
Hi @NilsRo thank you for your report.
I can reproduce this also on a Ultimaker S5 when using a 0.3mm profile. 
I will bring it up for discussion, keep you posted.

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  Can I close this?

closed 2024-12-09 by github-actions[bot]