Ultimaker/Cura #11687: Marketplace materials not appearing in available materials list

AndrewLindsay, 2022-03-20:
### Application Version

4.13.1

### Platform

Mac OS Monterey

### Printer

Creality Ender 5

### Reproduction steps

1. I added the Duport Hytrel material to my list of materials.
2. I rebooted Cura, and tried to select Hytrel, but it was not in the list of available materials.  I searched under all the different materials headings, in case it had been miscatagorized. 

### Actual results

I can not locate the Dupont Hytrel material as a material option in Cura, even though it appears in the material folder for the application.

### Expected results

I should be able to select the hytrel as the material for slicing my model.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]

fieldOfView, 2022-03-20:
The Marketplace only contains material profiles compatible with Ultimaker printers, and Ultimaker printers use 2.85 mm filament. Your printer does not support 2.85 mm filament. I know that all those materials are also available in 1.75 mm varaints, but the profiles on the Marketplace don't include this informaiton, let alone print settings for other printers than Ultimaker printers.

fvrmr, 2022-03-21:
Hi @AndrewLindsay thank you for your report. 
@fieldOfView is correct, see also: https://github.com/Ultimaker/Cura/issues/6236

I will close this issue, since it is a duplicate.

closed 2022-03-21 by fvrmr