Ultimaker/Cura #11688: Z hop for supports

Silithas, 2022-03-20:
### Is your feature request related to a problem?

For some reason, my printer always knocks into supports knocking them off the bed when i want to keep my supports minimal, but i can't when the nozzle constantly bumps into them either when traveling which i have set to avoid, yet don't or traveling to print on them.

So i want to request a z hop setting for z hopping onto it when the nozzle is XX millimeters away from support area.

### Describe the solution you'd like

Printer z hops onto the support start position when close enough

### Describe alternatives you've considered

IPrinting a more solid support with more walls, but which uses more plastic than wanted.

### Affected users and/or printers

Ender 3 pro, but with e3d V6 hotend and non original firmware/motherboard.

fvrmr, 2022-03-21:
Hi @Silithas thank you for your request.
I will bring it up with the team, keep you posted.

fvrmr, 2022-03-24:
I have discussed it with the team and we decided to defer this request, since it will not get a high priority on our backlog.
I hope you understand.

Silithas, 2022-03-25:
Yep, fully understand and hoping to see it as a feature for a later update if added to consideration in the future.

Do you know of a workaround i could try to not bump into the supports possibly knocking it over when traveling over/to the supports with what's already in cura?

Ghostkeeper, 2022-04-10:
Did you try disabling the Limit Support Retractions setting? When that is enabled, it won't retract any more when moving from support to support. And when it doesn't retract, it doesn't Z hop either.

closed 2023-07-13 by github-actions[bot]