Ultimaker/Cura #11694: running cura from source- pynest2d build problem

danielbarcohen, 2022-03-21:
### Application Version

4.9.0

### Platform

windos 10

### Printer

ultimaker s5

### Reproduction steps

im followig this guide:
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows
and while trying to do step 10. in pynest2D, the build fails.
at first, the python version was not right(the manual says download python 3.8.10, and the cmake demends 3.10, so i used the -DPython_VERSION="3.8.10" parameter.
still, fails, now saing somthing is wrong with the SIPMacros file(see actual results) 

### Actual results

-- Linking and building  against Python 3.8.10
-- Found SIP version: 4.19.25
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Found libnest2d: C:/dev/libnest2d/install_dir/include
-- Setting C++17 support with extensions off and standard required
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Found clipper: C:/dev/libnest2d/build/dependencies/include/polyclipping
-- Found NLopt: C:/dev/libnest2d/build/dependencies/include
-- SIP: Generating pyproject.toml
-- SIP: Generating source files
'C:/Program Files/CMake/bin/cmake.exe' '-E' 'env' 'PYTHONPATH=C:\Users\DanielBarCohen\AppData\Local\Programs\Python\Python38;C:/dev/pynest2d/build' '/Scripts/sip-build' '--pep484-pyi' '--no-protected-is-public'
The system cannot find the file specified
-- SIP: Touching the source files
-- SIP: Collecting the generated source files
-- SIP: Collecting the user specified source files
-- SIP: Linking the interface target against the shared library
-- SIP: Installing Python module and PEP 484 file in C:\Users\DanielBarCohen\AppData\Local\Programs\Python\Python38\Lib\site-packages
-- Configuring done
CMake Error at cmake/SIPMacros.cmake:56 (add_library):
  No SOURCES given to target: sip_pynest2d
Call Stack (most recent call first):
  CMakeLists.txt:54 (add_sip_module)

CMake Generate step failed.  Build files cannot be regenerated correctly.

### Expected results

i expect that it will build the pynest2d so i can use nmake and nmake install.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

.

jellespijker, 2022-03-24:
Hi @danielbarcohen you're running into this issue because we accidentally merged a development branch to master a bit prematurely. We are currently in the process of updating PyQt5 to PyQt6 and for this we need a newer version of sip 6.x to generate the Python bindings. Since Cura 4.13 will be the latest Cura version which uses PyQt5 and sip 4.19.x we decided to not undo this merge. Because all master branches from Cura, Uranium, libCharon, libSavitar and CuraEngine will be in sync anyhow in a couple of days.

I created a 4.13 branch from the last commit before the sip 6.x changes. If you checkout that branch you should be able to compile from source.

The commands below should do the trick
```bash
git pull
git checkout 4.13 
```

danielbarcohen, 2022-03-24:
> Hi @danielbarcohen you're running into this issue because we accidentally merged a development branch to master a bit prematurely. We are currently in the process of updating PyQt5 to PyQt6 and for this we need a newer version of sip 6.x to generate the Python bindings. Since Cura 4.13 will be the latest Cura version which uses PyQt5 and sip 4.19.x we decided to not undo this merge. Because all master branches from Cura, Uranium, libCharon, libSavitar and CuraEngine will be in sync anyhow in a couple of days.
> 
> I created a 4.13 branch from the last commit before the sip 6.x changes. If you checkout that branch you should be able to compile from source.
> 
> The commands below should do the trick
> 
> ```shell
> git pull
> git checkout 4.13 
> ```

can you elaborate please?
how should i follow the manual, which links should i change?
if you can explain which steps in the manual i should change and how, it would help me so much!

jellespijker, 2022-03-24:
Hi @Wikunia your issue seems to be a bit different then what @danielbarcohen is experiencing. It looks like you're linking against a wrong protobuf version. Sine protobuf isn't a dependency of pynest2d and libnest2d, I believe your issue is caused by libArcus (remember that CuraEngine also uses protobuf). It could be the case that the `find_package(Protobuf...` finds a system wide installed protobuf version. Make sure that you set the `CMAKE_PREFIX_PATH` to the root directory (contains the `bin`, `lib`, `share` etc. of your installed dependencies). If that doesn't help you could try set the Protobuf_\<VARIABLES\> manually https://cmake.org/cmake/help/latest/module/FindProtobuf.html

 If you're still seeing these linker errors you could try to investigate with: `ld libArcus.a` or `ld libArcus.so` and `ldd libArcus.a` or `ldd libArcus.so` to check which protobuf version it links to.

These issue will hopefully be a thing of the past in the near future. We're preparing to simplify how we handle dependencies. Once Cura 5.0 has been released we will put some effort into using Conan (https://conan.io/) as a dependency manager. Installing Cura from source should boil down to something along this line:

```bash
conan instal . -pr:b cura_build.jinja -pr:h cura_release.jinja --build=missing
```
This command should download, build, install and use all dependencies we use.

jellespijker, 2022-03-24:
@fvrmr FYI. I'm closing this issue since it isn't an actual bug in Cura.

closed 2022-03-24 by jellespijker