Ultimaker/Cura #11699: ULTIMAKER CURA FECHA AO TENTAR ADICIONAR OUTRO ARQUIVO

Cars118, 2022-03-22:
### Application Version

4.12

### Platform

WINDOWS 10

### Printer

CR-10S

### Reproduction steps

ULTIMAKER CURA FECHA AO TENTAR ADICIONAR OUTRO ARQUIVO

### Actual results

ULTIMAKER CURA FECHA AO TENTAR ADICIONAR OUTRO ARQUIVO

### Expected results

ULTIMAKER CURA FECHA AO TENTAR ADICIONAR OUTRO ARQUIVO

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

ULTIMAKER CURA FECHA AO TENTAR ADICIONAR OUTRO ARQUIVO

Cars118, 2022-03-22:
> Please be aware that this issue-queue uses the english language, and that
> by using all caps it looks like you are screaming. Because of the way you
> filled in the issue template, you are not coming across in way that
> inspires sympathy for your cause.
>
> Here's an article that may apply to your issue:
>
> https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file
> If it does not apply to you, then please provide more detail than one
> short sentence repeated 5 times.

Good afternoon,
I'm sorry, I didn't mean to shout, but I didn't know how to fill out
the form in the best way.
The problem that is happening to me is when I want to open several
files in Cura, it automatically closes the program, there will be a
solution for that.
thanks
Carlos Silva

Cars118, 2022-03-23:
> Do you have Chitubox installed on the same computer? If so, read the
> article I linked to before. If not, let's dig deeper.
> Let's try to make clear what is going wrong exactly.
> When you say you want to open several files, do you mean "several files *at
> the same time*", or "several files, *one at a time*"? Or do you just mean
> several specific files? Can you succesfully open one file in Cura? How are
> you opening the "several files"?
> Please locate the Cura.log file and post a link to it here. See

Good morning,
I appreciate the availability to help me,
 I uninstalled chitubox and it solved the problem and I can now open
several files at the same time.
 I just can't see the preview of the files in the files to know what object
it is.
 Is there any feature or update to get the preview to work?
Thanks,
Carlos Silva

fvrmr, 2022-03-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/9668 then.

About the preview you can install a stl thumnail viewer like this one: https://github.com/unlimitedbacon/stl-thumb

I will close this because it is a duplicate.

closed 2022-03-24 by fvrmr