Ultimaker/Cura #11700: Release the nozzle temperature limit

magicmaker3, 2022-03-23:
### Is your feature request related to a problem?

I need to print PEEK material at 420 ℃，But it reports an error, saying the setting is wrong。

### Describe the solution you'd like

Release the nozzle temperature limit

### Describe alternatives you've considered

Release the nozzle temperature limit

### Affected users and/or printers

Users who will make more breakthroughs

### Additional information & file uploads

[image]

fieldOfView, 2022-03-23:
In the `"overrides"` section of the printer-definition for your printer (.def.json file), you need to specify a new (higher) `"maximum_value"` for `"default_material_print_temperature"`, `"material_print_temperature"`, `"material_print_temperature_layer_0"`, `"material_initial_print_temperature"` and `"material_final_print_temperature"`.

fvrmr, 2022-03-24:
Hi @magicmaker3 thank you for your report.
Could you let us know if you can make it work with the solution from @fieldOfView?

closed 2022-04-07 by no-response[bot]