Ultimaker/Cura #11701: Retract Tower Script not working

Tempestine, 2022-03-23:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

1. Imported the Retract Tower part for calibration.
2. Added a post processing script "RetractTower" that was packaged with Cura with following settings:
Command - DIstance
Starting Value - 1
Value Increment - 1
Change Layer - 39
Change Layer Offset - 3
Display on LCD - Yes
3. Generated G-cod
[image]
e

### Actual results

Upon examination of G-code, the initial value and its increments are not calculated in the retractions. It appears to retract by the set amount in Cura (4.5mm
[ENDER_PETG Retraction Tower.zip](https://github.com/Ultimaker/Cura/files/8332865/ENDER_PETG.Retraction.Tower.zip)
)

### Expected results

The G-code should calculate the initial and incremental retraction settings and apply it to the calculations.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

N/A

GregValiant, 2022-03-23:
It does appear to use the Cura "Travel | Retraction Distance" up until the end of the "Change Layer Offset" layer.  From that point it starts to use the setting you noted and the retraction distance changes every 39 layers.  The following are from your posted gcode file:
;LAYER:4
M117 START distance (1.0/1.0)
Then later:
;LAYER:43
M117 retract E2.0
;LAYER:82
M117 retract E3.0
;LAYER:121
M117 retract E4.0
;LAYER:160
M117 retract E5.0
This is the last notice:
;LAYER:196
M117 retract E5.0
In the Gcode numbering layer 196 is the top layer (layer 197 in the Cura preview).
Looking at the retractions at each change, the distances are correct and continue until the next change.

The ToolTip for "Change Layer Offset" is not good.
The numbers that are input into the "Change Layer Offset" box and the "Change Layer" box should maybe be integers.  When I saw the decimal points I thought it wanted a height.

closed 2022-03-31 by fvrmr

Tempestine, 2022-10-11:
Good Morning,

Wow, thank you for the fast response. You are indeed correct about the
retraction distances. It was really late when I was sifting through the
lines...

Thank you for pointing out the interger issue also.

Will this retraction tower script work with a machine with 2
extruders/nozzles?

I have had issues in the past where a sliced print defaults to T0 instead
if T1 because of the sequence of priming lines. (T0 being the latter line
then all following code defaults to that extruder)

Kind Regards,
Paul

GregValiant, 2022-10-11:
You can try adding "T{initial_exttruder_nr} as the last line in your StartUp Gcode.  You may require a couple of other lines if the temperature is not the same or if T1 has other differences from T0.