Ultimaker/Cura #11702: Incorrect slicing

jmgk77, 2022-03-23:
### Application Version

4.13.1

### Platform

windows, linux

### Printer

ender 3, others

### Reproduction steps

Hi,

i got users reports that the file [borboleta2.stl](https://github.com/jmgk77/LAMPADA/blob/main/STL/borboleta2.stl) is wrong, that the thread is filled. The model is a FreeCad generated one.

[image]

Investigating, i found that Cura isnt generating correct gcode. In a [viewer](https://gcode.ws/):

[image]

In PrusaSlicer (or SuperSlicer):

[image]

### Actual results

File isnt sliced correctly

### Expected results

File should be sliced correctly

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_borboleta2.zip](https://github.com/Ultimaker/Cura/files/8335464/CE3_borboleta2.zip)

smartavionics, 2022-03-23:
The model isn't watertight (aka manifold) and needs fixing to be printable.

[image]

GregValiant, 2022-03-23:
The model on the left has been repaired.  There were a total of 1,530,079 errors in the model of which 673,037 were duplicate faces.  That is a very high resolution model and so the high number of errors.
[image]

Asterchades, 2022-03-24:
Slic3r and its derivatives (both SuperSlicer and PrusaSlicer) have built-in fixers that try to correct for common issues. They aren't perfect, though, and can create new and different issues of their own in some cases as they have to "guess" what the modeller actually wanted. There will be an indicator in the "Info" section in the lower right of the Platter view showing how many fixes have been applied, listed as "Auto-repaired **** errors".

Cura has a few automatic fixes built in which mostly default to being off (again because they can only "guess" as to what is correct), and there's a Mesh Tools plugin with some more.

fvrmr, 2022-03-24:
Closing this issue, since it is not an issue with Cura.

closed 2022-03-24 by fvrmr