Ultimaker/Cura #11714: CuraEngine build problem

Samuel111116, 2022-03-27:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Following [this guide](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows#curaengine), step 15 where you build and install CuraEngine, fails.

### Actual results

C:\dev\CuraEngine\cmake_build>mingw32-make
[  7%] Built target stb
Consolidate compiler generated dependencies of target clipper
[  8%] Built target clipper
CMakeFiles\_CuraEngine.dir\build.make:72: *** target pattern contains no '%'.  Stop.
mingw32-make[1]: *** [CMakeFiles\Makefile2:140: CMakeFiles/_CuraEngine.dir/all] Error 2
mingw32-make: *** [Makefile:155: all] Error 2

### Expected results

CuraEngine builds normally.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

danielbarcohen, 2022-03-30:
hey, how did you donlowd mingw?
their link to the donlowd dosent work at all

shenchumo, 2022-03-30:
I have the same problem

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  Can I close this?

closed 2024-12-09 by github-actions[bot]