Ultimaker/Cura #11716: [master]: UM3: Missing support

thopiekar, 2022-03-27:
### Application Version

master

### Platform

Linux

### Printer

UM 3

### Reproduction steps

* Load a (complex) model
* Slice it
* Review the result in the "Preview" tab

### Actual results

Support is missing or not even touching the critical contour.

### Expected results

Support is being added at whatever places are needed to get it printed well.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[UM3_FullAssembly.zip](https://github.com/Ultimaker/Cura/files/8357497/UM3_FullAssembly.zip)

GregValiant, 2022-03-27:
If that is the @smartavionics "Master" then you would need to report this [HERE](https://github.com/smartavionics/Cura/issues).
Slicing with my settings in Cura 4.13.1 I can't duplicate this (I can't duplicate it in Master either).  Use the "File | Save Project" command then zip up the file and post it over there.
[image]

thopiekar, 2022-03-27:
Strange. I could easily reproduce it by reopening my own project file.

GregValiant, 2022-03-27:
The file you posted is a 3mf file, but it is just a model file.  I think you used "Export" instead of "Save Project".  When I open the file my Ender stays active and it should be switching to a UM3.

thopiekar, 2022-03-27:
Good point. My mistake!
Used an older version of Cura to create this project file, but I can partly reproduce the errors with it when loading it into the master build here:
[image]
[image]
[UM3_FullAssembly.zip](https://github.com/Ultimaker/Cura/files/8358070/UM3_FullAssembly.zip)

fieldOfView, 2022-03-27:
Isn't this solved easily by lowering the `Minimum Support Area` value? (or increasing the `Support Horizontal Expansion`)

thopiekar, 2022-03-27:
Eventually, yes. But why are some other areas (which are much smaller) supported and the one in the back not?
Also, how does it come that the support in the front right is chopped and not supporting the tip?

Haven't tried, but don't understand how tuning `Minimum Support Area` would help here.. 🤷‍♂️

GregValiant, 2022-03-27:
I'm curious as to how you are going to get the support from under this shelf?
(BTW Is this to be mounted to a robot arm?  That round hole pattern sort of has that look to it.)
[image]

thopiekar, 2022-03-27:
Yeah, this design is a quick shot. I'm planning to make an assembly later with clamps, etc., but want to make a quick POC. Also, you will notice partly covered sinkholes. I will post-process it later and improve the design when the placement really makes sense.

@GregValiant You tried to reproduce my problem on Windows, too? Eventually, my bug is OS-dependant?

GregValiant, 2022-03-27:
It could be but I have a feeling it is more dependent on the way the support settings are configured.  Changing the Horizontal Expansion of the support will make those tall skinny structures more robust.  Matching the support HE to the support interface horizontal expansion will give the thin edges more to rest on.  Changing the support infill to lines or grid should get rid of that hole in the one support leg.
And if the support is PVA then no problem getting it out from under that shelf feature.

thopiekar, 2022-03-28:
Ok, well since you both are mentioning the setting, I guess I miss understanding its purpose or the impact it has.
However, I still don't understand why I get different results on symmetric models.

And the incomplete support in the front right
[image]

GregValiant, 2022-03-28:
But changing some settings it looks like this...
[image]

Here is a project file.  It has my Ender 3 Pro instead of your UM3 but the support settings should be apparent.  I added Z-hops because those tall skinny supports are prone to getting knocked over by the nozzle.
(If I was doing that particular model I'd make it three pieces and glue them together.  For a PoC you should be fine.)
[GV_UM3_FullAssembly.zip](https://github.com/Ultimaker/Cura/files/8366143/GV_UM3_FullAssembly.zip)

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  Can I close this?

closed 2024-12-09 by github-actions[bot]