Ultimaker/Cura #11719: Arachne engine can't use on Radeon GPU?

nunsamu, 2022-03-28:
### Application Version

Cura arachne engine beta 2

### Platform

Windows 10

### Printer

Flyingbear Ghost 5

### Reproduction steps

1. Upload the .stl file
2. Slicing
3. Preview

### Actual results

After slicing then go to preview, it freezing 99% of loading
BTW slicing is perfectly work. I just save the G-code file and upload on cura 4.13.1 to check for works
(Cura 4.13.1 version is works well(slicing, preview, and others))
[image]

### Expected results

It should show preview but doesn't D:

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I'm using RX6600 GPU
Previously I use Nvidia P400 GPU, It works well except slightly lags.

I know preview is only using GPU, so i suspect that Radeon GPU is not compatible on arachne engine. but I'm not sure, so I leave a report.
[cura.log](https://github.com/Ultimaker/Cura/files/8363320/cura.log)

SkyBeam, 2022-04-29:
I am using Cura 5.0.0-beta+1 now and it's working fine for me.
System: AMD Radeon RX 6900XT, AMD Software Adrenalin Edition 22.4.2, Windows 10

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If the problem persists into Cura 5.13.0 please fill out a new report so it goes to the front of the queue.

closed 2026-09-07 by GregValiant