Ultimaker/Cura #11722: saved .3mf projects do not contain any object file

christianpatrickgalles, 2022-03-28:
### Application Version

4.13

### Platform

windows 10

### Printer

CR10 mini

### Reproduction steps

I have been having this issue with all installs of cura on two different computers
I slice the project, then save project as .3mf.

### Actual results

No matter which order I do this the project files contains content (for example 60,000 kb worth of data) but Cura states "can not find object within C:/ location.

### Expected results

IT should have the object within the saved project!

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://drive.google.com/file/d/1k6BiyoT6Zue1uwVliGwgi25_TTECmClg/view?usp=sharing

christianpatrickgalles, 2022-03-28:
I've been repairing the .3mf files in micosoft 3d builder. they say the files are "invalidly defined"

fieldOfView, 2022-03-29:
Please share your project-file somewhere where we don't have to request access.

> I've been repairing the .3mf files in micosoft 3d builder. they say the files are "invalidly defined"

AFAIK, that message means the _model_ (not the file) needs repairing, which is why you loaded it into 3d builder in the first place, right?

fvrmr, 2022-03-31:
Hi @christianpatrickgalles thank you for your report.
Could you share your project file in a .zip here. Like @fieldOfView mentioned we can't access the file via the link.

closed 2022-04-14 by no-response[bot]