Ultimaker/Cura #11723: Getting T1 and T0 commands when printing with two extruders and one hotend

Dippenaarj, 2022-03-29:
### Application Version

4.10.0

### Platform

Windows 11

### Printer

Geetech A20M

### Reproduction steps

Download somthing like flexi hourse key chain 1 and 2
add both to print bed
select part 1 from Extruder 1 and part 2 from extruder 2

### Actual results

Printer stops when swapping from T0 to T1, seems to drop temp with a bit of a pause and then the print fails

### Expected results

to print from both extruders and 1 hotend

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[]
[GA_Flexi_Horse_Keychain_Part_1.zip](https://github.com/Ultimaker/Cura/files/8367499/GA_Flexi_Horse_Keychain_Part_1.zip)
(url)

GregValiant, 2022-03-29:
The Gcode file looks good.  There are only four M104 lines and two M109 lines in the file.  They all set the temperature to 205 (except of course the last M104 S0 that shuts down the hot end).
There are 65 tool changes in the file and it appears you have properly set "Extruders share heater" and "Extruders share nozzle" in the Printer Settings.  These are from the settings at the end of the file:
[values]
machine_extruders_share_heater = True
machine_extruders_share_nozzle = True
material_final_print_temperature = 205.0
material_initial_print_temperature = 205.0
material_standby_temperature = 185.0  **<--- This temperature doesn't appear in the gcode proper**
retraction_amount = 9.0  **<--- That is a very long retraction distance**
switch_extruder_retraction_amount = 18.0

The tools are being called and the temperature is not bouncing around in the gcode.  Loading the gcode into Cura the preview looks good and reading the file into AutoCad the gcode looks good.  Right now I'm thinking the problem may be in the printer.  Have you tried printing anything with Extruder 1 disabled and using only Extruder 2 (T1)?

Dippenaarj, 2022-03-29:
Hi,
Thank you for the reply, I'll try and print it via the SD card, it looks like the issue is with Octoprint and not with Cura, thank you for confirming that for me.

closed 2022-04-14 by no-response[bot]