Ultimaker/Cura #11726: No support for CR-5 Pro H

bridgettegraham, 2022-03-29:
### Application Version

4.13

### Platform

Windows 10

### Printer

CR-6

### Reproduction steps

Please can you add support for CR-5 Pro H

### Actual results

there is currently no support for CR-5 Pro H

### Expected results

Please can you add support for CR-5 Pro H

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info

GregValiant, 2022-03-29:
The definition files come from the manufacturer (Creality in your case) or a community member.  They aren't updated between Cura versions.  Is there another Creality printer that is close?

GregValiant, 2024-11-15:
No one as added a pull request for this printer.  I'll go ahead and close this.

closed 2024-11-15 by GregValiant